kevancress/MeasureIt_ARCH #136: Dimension between objects

meutzitzu, 2020-10-05:
Since Blender 2.8 allous us to edit multiple objects at the same time, why can't MeaureIt-ARCH draft dimensions between 2 arbitrary vertices, regardless of what object they come from ? 

I think this is definitley a much needed feature. For example, when assigning dimensions for windows and doors along walls, I have the walls as a bunch of edges on the XY plane, that are then extruded vertically using a screw modifier with 0 angle, and then thickness added to them using a solidify modifier. the windows are just an object with the exact same configuration, but offset vertically, and with higher thickness on the solidify, which is then subtracted from the walls object via a Boolean difference. 

Using this workflow I can't spedify the dimensions to any of the windows/doors because I need to refference a vertex from the walls object, and one from the doors/windows object. This should definitely get fixed, because there's absolutely no way in hell I'm going to apply all those modifiers. My current workaround is to add a plane object, intersect it with the walls, apply that boolean and dimension everything. If I have to change the design (which happens a lot) I'm just going to have to replace the intersected plane with a different one and dimension everything from scratch again.

kevancress, 2020-10-05:
Hey @meutzitzu, MeasureIt-ARCH can actually do measurements between vertices of two different objects, It just can't define it in edit mode (since it needs to know the transform matrix of the object the vertex belongs two in order to work out its world space location) but if you select the vertices in multi object edit mode, then switch to object mode you can apply the dimension then.

Here's a quick demo with the setup you described (also I'd never thought about using the screw modifier that way, that's a really nice workflow!!!):
[image]

I know this workflow isn't really ideal, it's a pain to have to keep switching modes. I'll see if I can find a way to keep track of a vertex's parent object in edit mode to make the workflow better! another approach would be to dimension between two empty objects, and use vertex parenting to keep the empties attached to the geometry.

meutzitzu, 2020-10-16:
Even when switching modes it still doesn't work for me ... for some reason it keeps adding the dimension between the objects' origins 
[M_ARCH_test.txt](https://github.com/kevancress/MeasureIt_ARCH/files/5391322/M_ARCH_test.txt)
rename from .txt to .blend, obviously 

I'm using the latest master (downloaded just 30 mins ago)

is there something I'm doing wrong ? should I upgrade to the version_5 branch ?

kevancress, 2020-10-17:
Hmm, Version 5 has some major changes that will break older files, so I wouldn't switch if you're in the middle of working on anything major.

I tested out the file and it seems to be working fine for me in both master and the version 5 branch.

There needs to be only one vertex selected in each object before you switch back to object mode. And that vertex has to be the active vertex (white highlight not orange) so using circle or box select won't work.

You can also overwrite the vertex manually in the measureit_arch data... It's a bit of a pain but here's a video of how to do it.

[image]

... yeah I definitely need to make this an easier process.

meutzitzu, 2020-10-18:
jesus christ ... talk about "under the hood" 
I had no idea you could acces all that data in the outliner

meutzitzu, 2020-10-19:
okay, it now works for me 
This is great!
 ... it was just me being too used to box select lol.

kevancress, 2020-11-10:
Okay!! I think I finally managed to get this working properly all in edit mode!!!

Just committed the fixes in the master and version 0.5 branches! Here's a quick demo:

[image]

Let me know if you find any issues with it!

kevancress, 2020-11-10:
Gonna close this for now, but open up a new issues if there's any problems!

closed 2020-11-10 by kevancress