kevancress/MeasureIt_ARCH #138: partial dependency-graph evaluation

meutzitzu, 2020-10-18:
I've been thinking ... So I know how the deformative mesh modifiers work A-OK with M-it_ARCH, but the second you try to include generative modifiers everything gets screwed up ... 

You usually don't want the dimensions to take into account the complex mesh that comes out of the modifiers, but instead only refference the vertices themselves which you can acces in edit mode.

that being said, sometimes it is useful to  constrain multiple vertices relative to eachother or, rather to a refference object such as an empty, which you do by hooking them to said empty. 

This doesn't happen often and I have no idea how feasable it is, if at all, but I've been thinking of evaluating the modifiers as they are in edit mode (if those show in edit and show on cage mesh icons are checked on the modifiers)

kevancress, 2020-10-18:
hmm, yeah that's a good point, I hadn't really considered how those generated verts are stored and displayed for the cage mesh and show in edit mode buttons on the modifier. I'll do some digging in the API, and see if there's any way to consistently access/ reference them... I need to dig into the depsgraph stuff some more anyway to fix how the line groups are getting cached anyway (it's a little broken right now), so hopefully I'll find a way to do it!

kevancress, 2020-11-10:
Okay so I stumbled on this accidentally while working on another project, but. It turns out that with the very under the hood method of directly adjusting the vertex indices that the dimension is connected to, this is, in fact, possible (as long as the evaluate depsgraph checkmark is ticked on the dimension.

Demo Gif:
[image]

Sooo The hurdles to get this to be an actually supported feature are:

1. How do I deal with all the bugs (looks like the generated vert indices aren't super consistent, so I'll have to sort that out)
2. I need to tweak my garbage collection system so it doesn't immediately delete these (right now it just assumes it's a dimension with invalid indices whenever you enter edit mode, so it deletes it...)
3. How do I make it easy for a user to connect a dimension to an evaluated index, if it doesn't show up in edit mode?? 

Anyway, at least we know its possible!

meutzitzu, 2020-11-11:
_Anyway, at least we know its possible!_ Welp, that sounds pretty good, ngl