kevancress/MeasureIt_ARCH #139: seems buggy

Shaddy-ElDesouky, 2020-10-22:
[image]
this report while in edit mode, selecting one vertex, default cube, trying to add annotate.
Also line preview doesn't show in edit mode but shows normaly after I switch to object mode, while dimension shows normaly in edit mode.

MeasurIt_Arch 0.4.6
Blender 2.90.1
Linux lite  (ubuntu 18.0.4 derivative)
Lenovo thinkstation S20 xeon x5650 8gb ecc
gtx 1050ti

Note: MeasurIt_Arch 0.5 doesn't let me check the box in the addon manager

kevancress, 2020-10-25:
Hey @Shaddy-ElDesouky Thanks for the bug report. Measureit-ARCH is still very much in development, so it is pretty buggy.

I haven't been able to reproduce the issue you're reporting on windows 10 or Linux (PopOS). From the looks of the error message and your description it's a problem with Blender trying to call the add empty operator from in edit mode, with incorrect context, are you using the add annotation button in the toolbar, or calling the operator from another script?

Maybe try updating your Blender Version and let me know if that fixes it.

For the version 0.5 activation issue, can you post the error message that occurs when you try to activate the addon?

Shaddy-ElDesouky, 2020-11-20:
Hello @kevancress , v 0.5 activation issue , when you try to install the addon it installs but when you try to check the box to make it active the box is not checked.
[image]

kevancress, 2020-11-20:
Ahhh okay thanks the error reports help! It's an issue with how GitHub names the downloaded zip package. I've renamed the branch which should fix the problem. Remove the version that won't enable and give the new https://github.com/kevancress/MeasureIt_ARCH/tree/version_0_5 branch a try, it shouldn't have the naming issue.

Shaddy-ElDesouky, 2020-11-21:
@kevancress  _0_5 worked :+1:  , I don't like to bother you but do you want me to report every bug I find, If that will help I will try to do this, also I'm not experienced with bug reporting but what I understand that each issue should be reported separately, at the first glance in Dimension the dashed scale is very big compared to triangle, even scaling it down to 1 instead of 15 (default) it is still big and there is no decimal so I could scale it less than 1.
[image]
[image]

Shaddy-ElDesouky, 2020-11-21:
tried the main issue with V _0_5  Changed the OS but still the same problem.
OS: Kubuntu 20.04.1 Blender 2.90.1
[image]

kevancress, 2021-03-04:
@Shaddy-ElDesouky Sorry I completely forgot to post here after add the fixes for both of these issue! But both are fixed in the latest development branch now!

closed 2021-03-04 by kevancress