kevancress/MeasureIt_ARCH #140: Accurate placement via Gizmos

meutzitzu, 2020-10-24:
I have OCD, I know ... big surprise. I can't sleep at night properly if I know my dimensions are just manually tweaked so they look like they're aligned, but if you look in the Distance field you see only floating point gore. 

To fix this, ( and this *does* happen a lot ... when you have multiple colinear dimensions both ISO-129 and common sense dictates that the dimension segments should share the same line ) I propose 2 things : 
1) that the CTRL manipulation modifier works with the gizmos too (SHIFT already does it's thing ... if you hold it down, it moves the gizmo slower, so holding down CTRL should move it in increments)
2) that we have an N-panel toggle for whether or not the add-on should add that original Distance to the dimension which is somehow proportional to the magnitude of the dimension. simply add the dimension exactly on the edge ( with initial distance of 0 )

kevancress, 2020-10-25:
Aha yes I know what you mean, I was exporting some drawings to .svg last week and was driving myself crazy because some dim's were just a hair off of being correctly aligned!!!

Getting the CTRL modifier working would be fantastic. I've been neglecting the gizmo's a bit lately because the API for them hasn't been super well documented. But I'll circle back to it and see if I can get that working!

kevancress, 2020-10-25:
Still a bit buggy, but I've got a start on this in master eb7e0538eef8b531ccd6640370655c485e386208. For some reason the gizmos are always drawing at the base of dimension now, not sure why but I'll keep working on it.

Also the snapping isn't taking units into account yet, so It's fine for metric, but imperial is a bit broken...

kevancress, 2020-11-02:
Added a fix for imperial snapping and forgot to post about it here. It's in 37ffe69d55c0c7a3e62a185d934b3822135de67f (master and v0.5) 

Gonna close this for now, let me know if there's any issues though!

closed 2020-11-02 by kevancress