kevancress/MeasureIt_ARCH #144: Vertical text and items selection with viewport

irvnriir, 2020-11-12:
I didn't find a way to make it keep text (only) aligned to the camera rotation (to right, instead of up/down) .

And the feature why i look beyond the default Measureit, is an ability to delete the measurements trough any kind of selection in Viewport (for example selecting the primitives) . Because when there is 50+ of them, searching the lists is absurd . This feature seems to be missed here too .

kevancress, 2020-11-12:
Hey @irvnriir 

Your right, MeasureIt-ARCH dimensions and annotations don't have a way to keep text aligned to the camera yet. I'm working on adding this to annotations soon.

For deleting dimensions: there's no way to delete them directly in the viewport. But using the dimension gizmo in the viewport will select that dimension in the list, so you don't have to search through them all! Here's a demo:
[image]

If it's not causing the dimension to be selected in the list, make sure you're using the latest version of the MeasureIt-ARCH master branch

irvnriir, 2020-11-13:
@kevancress good to know, though i probably won't switch, until Dimensions are aligned :)

kevancress, 2021-05-15:
@irvnriir I'm closing this issue for now, I've added an options for Annotations to stay aligned to the camera view. It's in the development branch now.

closed 2021-05-15 by kevancress