kevancress/MeasureIt_ARCH #145: Empty render output

dtrauner, 2021-01-08:
I am able to produce output for a simple blender file, but the output file is empty for a larger / more complicated file. I receive no errors and nothing is output to the console. Any recommendation for how to debug/track down would be great.

Thanks for the great addon,

kevancress, 2021-01-08:
Hey @dtrauner

What version of MeasureIt-ARCH are you using? If I'm remembering right, in the older versions (v0.3, v0.40) the restrict render toggle wasn't properly accounted for in the MeasureIt-ARCH render, so if you had a hidden object occluding MeasureIt-ARCH elements you'd see them in the viewport but the render would be blank.

I'd go through and move any potentially occluding objects into a collection and disable that collection in the view layer your rendering as a first step to see if that's causing the issue.

If that's not the issue, let me know! if your able to send the file over I'm happy to take a look and see if I help track down the issue!

dtrauner, 2021-01-08:
I am using Version 0.4.5. I have disabled most objects. I see the dimensions in the viewport, but the file output is a blank image. The console indicates a successful render "/mit_frame0004.png saved". 

It is a large file with many links. Any other debugging suggestions?

Thanks again.

dtrauner, 2021-01-08:
I have added some debugging info from measureit_arch_render.py.

at approximately line 297:

      sheetGen = myobj.SheetGenerator
      for sheet_view in sheetGen.sheet_views:
                    draw_sheet_views(context,myobj,sheetGen,sheet_view,mat)

Many items arrive at this point, but there appear to be no sheet_views. 

I added the plug in recently after the modeling was mostly completed.

Thanks

dtrauner, 2021-01-08:
Also around line 330

for obj_int in deps.object_instances:
            if obj_int.is_instance:
                myobj = obj_int.object

No objects are returning is_instance

kevancress, 2021-01-08:
hmmm okay, a couple more thoughts...

Are the dimensions attached to the objects linked from another file (or a collection being instanced within the file)? if so, you need to go the MeasureIt-ARCH settings (scene tab in the properties panel) and enable the 'instance dimension' checkbox. Dimension instancing is disabled by default because the dimensions can't update to local scaling and rotation of instanced objects, so they could display incorrect measurements. 

Sheets are an unfinished new feature, so they're shouldn't be any, that's to be expected. Same for the Instances, If the dimensions aren't attached to the instanced objects nothing should be drawn there. If you have a debugger running the place you want to look for problems is the draw3d_loop() function in measureit_arch_geometry.py, its line 3548 in the latest master.

also might be worth install the latest master too just to see if that fixes the issue, it should show v0.4.6(git) in the toolbar when its installed.

kevancress, 2021-01-08:
also just for reference, what operating system and graphics card are you running?

dtrauner, 2021-01-08:
Brilliant. 0.4.6 solved the problem. 

I am on OS x. 

Thanks so much.

kevancress, 2021-01-08:
ah fantastic! I'll close this for now then. Let me know if there's any other issues!

closed 2021-01-08 by kevancress