kevancress/MeasureIt_ARCH #147: SVG Export

mjepronk, 2021-01-18:
Hi. First, thanks for this Blender addon!

I saw you've added vector output in the latest git master branch (I don't think this is released yet). I was wondering if you would be interested in using FreeStyle SVG export for rendering the scene (only if this addon is enabled by the user), instead of a PNG render.

I've already build a rough prototype of the and it seems to work fine. Would you be interested in a PR with a more polished version?

kevancress, 2021-01-19:
Hey @mjepronk

That sounds Fantastic. Right now I'm handling svg creation with the svgwrite python library currently, and I've been meaning to look into integrating FreeStyle SVG to handel drawing more complex objects, but haven't gotten around to it yet, so if you've got a prototype working that would be amazing!

I'd definitely be interested in taking a look at a pull request!

mjepronk, 2021-01-22:
Thanks for merging this. I'm not sure if I would opt to add more settings, it tends to clutter the interface and make it more complicated for first time or non-regular users. For my part "embed scene render" and "embed freestyle svg" are always desired, and don't need to be an option. What was your use-case?

Alternatively, maybe we could chose some sane defaults and move most settings to a popup window or something.

kevancress, 2021-02-13:
Sorry it took so long for me to reply to this,

I've been using "embed scene render" and "embed freestyle svg" separately quite frequently, typically I'm using freestyle vector linework with MeasureIt_ARCH when I'm preparing technical drawings for Permit or Engineering reviews, while generally I only embed the raster rendering as an underlay when I'm producing presentation / client drawings. So It's nice to have them separate.

I do agree that the Rendering UI currently is a little messy/cluttered though... Maybe instead of the checkboxes we make it a dropdown....

mjepronk, 2021-02-14:
Hi Kevan, No worries, thank you for your time in reviewing and merging my PR's!
I'm going to think about ways to cleanup the UI, I'll create a separate issue when I've written it down.

closed 2021-02-14 by mjepronk