kevancress/MeasureIt_ARCH #149: Preset paper sizes

mjepronk, 2021-01-20:
I think it would be nice to have some presets available when specifying paper sizes in the view settings.

A dropdown with:
Custom (specify width / height)
US letter
A4 portrait
A4 landscape
etc.

Ideally, width / length would be disabled when not "Custom".

I saw presets/paper/{letter.py,tabloid.py}, but I'm not sure if they're used. However, why not just statically create a list:

```python
PAPER_SIZES = [
  ("US Letter", l("8.5in"), l("11in")),
  ("A4 Portrait", l("21cm"), l("29.7cm")),
  # ...
]

def l(str):
    # Could also use IMPERIAL
    return bpy.utils.units.to_value("METRIC", "LENGTH", str)
```

Would you like a PR for this?

kevancress, 2021-01-24:
Hey @mjepronk 

Yeah this would be great! the view system was based on my old [per camera resolution](https://github.com/kevancress/per_camera_resolution) addon, which had an attempt at presets, but they were always pretty rough. Lots of the old code to implement them is still hanging out in the 'custom_preset_base.py' file if you want to take a shot at getting it up and running again!

kevancress, 2021-02-13:
This works great!

I've adjusted the UI slightly, so that the width and height properties are still visible (just disabled) when using a preset, so that users who aren't as familiar with the sizes can still see what the dimensions are.

[image]

closed 2021-02-13 by kevancress