kevancress/MeasureIt_ARCH #152: Dimensions, lines, and annotations do not show

anthonyesau, 2021-02-08:
With the latest commit of the script (a9e2c8a9775ea428a8a8fde348cfbb34c207f535) none of the MeasureIt_ARCH features are visible even after toggling the show/hide button. This problem is for a file that was set up with a previous commit (not sure exactly which though). It appears that the one prior (9fd3fcb000ac7be053b81dc11b05aa4becb30c62) does still function as expected. 

Starting with a totally fresh file, new lines and annotations that are added do show up, but not dimensions.

macOS Catalina 10.15.7
Blender 2.90.1 and Blender 2.91.2

mjepronk, 2021-02-08:
Yeah, I've got the same issue. @kevancress here is a traceback of the error:

```
Traceback (most recent call last):
  File "/home/mp/.config/blender/2.91/scripts/addons/MeasureIt_ARCH/measureit_arch_main.py", line 639, in draw_callback_3d
    draw_main_3d(context)
  File "/home/mp/.config/blender/2.91/scripts/addons/MeasureIt_ARCH/measureit_arch_main.py", line 594, in draw_main_3d
    draw3d_loop(context, objlist)
  File "/home/mp/.config/blender/2.91/scripts/addons/MeasureIt_ARCH/measureit_arch_geometry.py", line 3630, in draw3d_loop
    context, myobj, DimGen, alignedDim, svg=svg)
  File "/home/mp/.config/blender/2.91/scripts/addons/MeasureIt_ARCH/measureit_arch_geometry.py", line 555, in draw_alignedDimension
    dim, dimProps, origin, dist, distVector, offsetDistance, capSize)
  File "/home/mp/.config/blender/2.91/scripts/addons/MeasureIt_ARCH/measureit_arch_geometry.py", line 3432, in dim_text_placement
    cardX = square[3] - square[0]
TypeError: unsupported operand type(s) for -: 'tuple' and 'tuple'
```

mjepronk, 2021-02-09:
@kevancress: I've fixed this in my PR with cosmetic changes.

There was a redundant cast from Vector to Tuple.

kevancress, 2021-02-13:
Hey @anthonyesau

Just pushed a fix for this in master based on the @mjepronk PR #151 

@mjepronk Thanks for all the work you've been putting into these Pull Requests!! Sorry I haven't had a chance to review them yet, I've been pretty swamped this past week, I'll try and review and merge them this weekend!!!

closed 2021-02-13 by kevancress

anthonyesau, 2021-02-18:
It's working great. Thanks @mjepronk and @kevancress!

commit 03790385 (2021-02-13): Fix #152