kevancress/MeasureIt_ARCH #154: Dimension style presets

mjepronk, 2021-02-12:
Hi Kevan,

I was thinking of automatically creating a set (about 3-5) of dimension styles in the .blend if there aren't ones defined yet. We can then also automatically set a default dimension style. Would you be interested in this? If so, I'll code up a PR.

If we chose to take this route, we may even deprecate dimensions without an associated style. I think we could simplify a lot of code in that case.

Let me know what you think!

kevancress, 2021-02-13:
Hey @mjepronk 

I really like the idea of preset dimension styles, with an automatic default style.

I think it's still good to keep the option to have unique Dims without having to set up a style, I use unique Dimensions quite often in my own projects.

It would be nice to simplify the code, but my first principle here is maintain user flexibility!

mjepronk, 2021-02-14:
OK, I'll create a PR for preset dimension styles in the coming or next week.

closed 2021-02-22 by mjepronk