kevancress/MeasureIt_ARCH #159: View output path and date folder ignored

mjepronk, 2021-02-22:
Hi Kevan,
I've just noticed that we don't use the `output_path` and `date_folder` of the view anymore.
If you like I can create a PR, but how do we want the functionality to work?
In `render_main` and `render_main_svg`, if the current view (result of `get_view()`) has an `output_path` set, we use that one instead of `scene.render.filepath`?
Optionally, if `date_folder` is checked we add the date to the path?
I also saw `output_path` on schedules. I haven't used schedules yet, and am unsure what it is, but should we check that one too?

kevancress, 2021-02-22:
Hey @mjepronk,

Actually, making a view active already sets scene.render.filepath to be the views output path.

It's in the update method in measureit_arch_views.py which gets called each time the viewContainers active index is changed.

Although setting it at render time like you're doing here is probably clearer / more efficient. Changing views can be pretty slow right now, and moving some things out of the update function like this might help speed things up!

I'll try and review and test this out this week!

mjepronk, 2021-02-23:
Aah, OK, I missed the fact that it automatically updated... so I think this behavior is more intuitive.
Bonus is the speed and we don't mess with the users' render settings.
No worry, take your time.

commit 214d6554 (2021-03-01): Merge pull request #160 from mjepronk/development

Render to view / date path when defined on view, fixes #159
commit cb879ccd (2021-03-01): Render to view / date path when defined on view, fixes #159
closed 2021-03-01 by kevancress