kevancress/MeasureIt_ARCH #161: Code refactoring

mjepronk, 2021-02-22:
Hi Kevan,

I've a proposal for another "cosmetic" change:
Since we're breaking compatibility with the latest 0.4.6 release now, I was thinking if this is the right moment to fix the inconsistency in property names on classes? We now have a mix of camelCase and snake_case. Personally, I would prefer snake_case everywhere (more Pythonic), but of course, the choice is up to you.

Another thing:
Would you be willing to accept PR's to refactor the dimension code to become more object-oriented? I'm playing with code to allow the user to select dimensions in the 3D viewport. But the problem is, right now it's very hard to find the coordinates of the text labels for example for a particular dimension. So I propose something like:

```
class AlignedDimension(BaseDim, PropertyGroup):
    dimObjectA: PointerProperty(type=Object)
    dimObjectB: PointerProperty(type=Object)

    def draw(self, context):
        # Basically: draw_alignedDimension(context, self, ...)

    def get_text_coords(self):
        ...

    ...
```

Hope that makes sense...

kevancress, 2021-02-22:
Hey!

Yeah I have a bad habit of swapping between camel and snake case somewhat arbitrarily, and you're right now is definitely the time to fix it! I'm fine with going with snake_case if that's more conventional for python code!

I'm definitely open to more Object oriented code too, I can certainly see some advantages to it. I think I'd steered away from it partly to stay close to the structure of the original MeasureIt addon (which MeasureIt-ARCH is based on) but I don't think that's really a priority.

Also I remember there being some issues with proper inheritance when Blender is loading addons, for example a draw function defined in BaseDim wouldn't be inherited by AlignedDimension. Worth taking another look at though for sure, but might be good to start with some smaller chunks to make sure that it does behave as expected?

Might be worth working on both these changes in a new branch?

mjepronk, 2021-02-23:
Yeah, I'll start a new branch and change properties to snake_case.

I will also try out refactoring the code to be more OOP style.  If that doesn't work out, I'll probably opt to refactor the code to smaller functions that could be reused for other functionality (like my use-case to select dimensions from the 3D view).

kevancress, 2021-02-25:
That sound great!

Yeah I think there's a lot of repeated (or nearly repeated) code across the various dimension drawing and creation functions that could definitely use a lot of clean-up.

I should probably put together a demo .blend file that we can use as a baseline to test against when refactoring to make sure dimension drawing and positioning behavior stays relatively consistent.

mjepronk, 2021-02-26:
> I should probably put together a demo .blend file that we can use as a baseline to test against when refactoring to make sure dimension drawing and positioning behavior stays relatively consistent.

If you could add a .blend file to the repository that would be great. Something like the simple house drawing from the README would be excellent. I then could easily create automated tests (like I did for the units module) that confirm that the dimensions show the expected values!