kevancress/MeasureIt_ARCH #166: SVG output font family

xyleme, 2021-03-23:
great piece of work Kevan!
Playing with SVG export, I stumble on the following issue: whatever custom font I set in a Dimension Settings, the outup vector file is composed with OpenSans, i.e. each text fill instruction states font-family="OpenSans Regular"

kevancress, 2021-03-24:
Hey @xyleme, 

Thanks for the reminder on this, I'd been avoiding doing proper vector fonts for a while since it required an external library. So I'd had the font family hardcoded as "OpenSans"

Just added support for this in 58de4877e44bf156309aa12e919243556e13f9df of the development branch, test it out and let me know if it works for you!

kevancress, 2021-03-24:
one more fix for this in 265bd9d7a963cb485e0319081b94a4749a29862f

xyleme, 2021-03-24:
Nice! thank you Kevan.
I noticed some font shapes don't export and sometimes mess up a dimension style. I guess it's a matter of original font files types: looks like opentype glyphs work like a charm whereas truetype fail. Anyway, it's so simple to locally tweak a svg file from within Inkscape...

kevancress, 2021-03-24:
Ahh that's good to know! I'll reopen this noting that some font file types still fail.

It seems for me that both .oft and .ttf font files work as long as they're installed on my system. Uninstalled fonts will show up in the viewport and raster renders but not in an svg export.

If you have an example of a font file that's causing issues and can share it that would be helpful in troubleshooting this! Also what Operating system are you running?

xyleme, 2021-03-25:
I run blender from a debian 10 system.
I tested several cases (locally vs system-wide installed fonts, otf vs ttf) but it still outputs "OpenSans Regular" font family. 
I noticed something that may be related: some font faces don't display; for example if you set font to "FiraSans-Light" it works, but with "FiraSansCondensed-ExtraLightItalic" then texts disappear. I mention the issue is not related to blender because both these faces display correctly in an ordinary blender text object. Hope this helps you...

For the time being, as my setup only uses one font style I just have to tweak the output svg file with sed:
cat View\ 1_0001.svg | sed s/OpenSans\ Regular/Fira\ Sans/ > View2.svg

kevancress, 2021-03-25:
That's really helpful thanks!

I've been testing on Windows 10, so it might be OS related. I've got an install of Pop_OS set up too so I'll give it a shot there and see if I can reproduce the issue a Linux OS. I'll let you know if I can resolve it

xyleme, 2021-03-25:
Ok, nothing especially blocking for me right now. I plan to use your add-on on a regular basis, so I'll probably send some pull requests soon.
Congrats for your hard work, I guess many engineers/architects/technicians do appreciate these capabilities finally added to blender.

closed 2021-03-25 by xyleme