kevancress/MeasureIt_ARCH #168: Animation transition is not smooth

18883234348, 2021-03-31:
[image]
Font Size and Arrow Size Animation transition is not smooth

kevancress, 2021-03-31:
Hey @18883234348  Yep this is a known issue,
Font size and Arrow size are both integer properties, so they can't have smooth animation curves. They could be changed to float properties, but that makes them a bit more cumbersome to set for common use cases.

kevancress, 2021-05-15:
Hey @18883234348  I've found a way to get measureIt-ARCH to treat font size and arrow size as floats for animations, but still display as ints in the UI, If you download the latest development branch it should fix your issue .

Fixed in: abfd7eba24d10478c112ad67b151b5c8122e8ad8 f34bf70d65f85b64c0228a2d700547acccec05de

closed 2021-05-15 by kevancress