kevancress/MeasureIt_ARCH #170: Object mode is doubling the dimension

PeterEller, 2021-04-13:
it's working fine in edit mode but in object mode when I add an aligned dimension between two objects it doubles the value and draws it in the wrong location 

I tried the same task with the basic measureit add-on and it works fine 
I added some screenshots to show the difference

 I'm on mac V 11.2.3 big Sur
 
[image]
[image]

kevancress, 2021-04-17:
Hi @PeterEller 

Thanks for catching this, I've just added a fix to the main development branch, if you download it dimesnions in object mode should behave as expected.

Also, for more precision you can add dimensions between objects in when in edit mode with multiple objects, this will use the selected vertices rather than defaulting to the objects origin point!

[image]

closed 2021-04-17 by kevancress

PeterEller, 2021-04-17:
thanks for the update
Yes I discovered that selecting both objects then going into edit mode allows me to select to vertices  for the align dimension 

Your add-on has been super helpful for my cabinet work!
Will happily pay for it when you have a pro version!

Cheers!

Peter Eller  
MODELLER MODERN SHELF SYSTEM
Direct: +1 (604) 736-5797 | ***@***.*** ***@***.***>
1157 East 23rd Avenue Vancouver, BC, V5V 1Y8
www.modeller.ca <http://www.modeller.ca/>