kevancress/MeasureIt_ARCH #172: MeasureIt_ARCH

Blendersbrontobyte, 2021-04-25:
Really Cool program works well. I use Inches.  This has created a problem.  Blender wants me to scale my drawing to show inches.  When I do that everything is totally messed up cant zoom or view right the grid is way off. The other problem that I am having is this 
 
[image]
This object is 8.003 long exactly.  I need measurit arch to use its precision as blender is very precise.  Thank you for adding on to the addon and making it better.  Wish I could find a way to make inches show up without scaling my drawing and I wish it woudl show up in ether fraction or decimal or metric i need it to show the precision.  Nott sure why its not.

Blendersbrontobyte, 2021-04-25:
[image]
so if I use metric it will show up but now I need to get rid of them M as my units are actually going to be scene by the rest of the 3d world as inches.  So very glad you allow me to keep the units hidden. 
[image]
In a perfect word I would be able to show this very same thing as 8.015625 as ether or 8 1/64th with measurit and blenders camera and rendering would know that my object is actually 60" not 60 meters.  Autocad believes its 60 inches even if I say its meters. When I export to dxf using units of m or none the object will pull into any cad program will look at those units as if they were inches.  This make the world go around for me. so hope THAT never changes.  
[image]
In a perfect world I could tell blender my object is 60" not 60 meters.   ALSO when I use the EEVEE express addon measurit arch does not get along with the rendering size due to something with camera scaling.   
 Thanks for the addon.    Hope to see it on blender Market or Gumroad.

kevancress, 2021-04-25:
Hey @Blendersbrontobyte 

I've just added a couple options that should help you out. You can get them by downloading and installing the development branch of MeasureIt_ARCH. It's not really a solution I'd recommend for future projects (more on that below) but should get things working for your current project without requiring any scaling of the objects. Here's a quick demo:

[image]

This adjusts the unit scale to redefine Blender base units in inches (using the 0.0254 scale factor). I wouldn't generally recommend this, but it does work.

For future projects, the key thing to remember is that, unless you adjust the unit scale like above, Blender always treats its base unit as defined in Meters, and swapping to the imperial units doesn't redefine those base units to be feet or inches (like a lot of traditional CAD tools do) but instead just changes the units being displayed. Personally I really like this method, as I think its a bit more real world accurate, i.e.. an object that's 2m x 8m would actually be ~6'7" x ~26' 4" , because changes in measurement units don't change the scale of the world... As a designer who has to provide documents in both metric and imperial this can be quite handy too, as I can produce accurate drawings in both measurement systems without having to scale the model.

When working in Imperial though its best to swap the units to imperial right at the beginning of the project, that will allow you to enter measurements in feet and inches right from the start, which will prevent problems down the line.

This all can make exporting to traditional CAD a bit tricky though, you'll have to test out what works best for your software, but generally I find exporting from Blender with units set to metric, then importing as Metric to CAD and adjusting units from there as needed to work best!

Also for scaling, if you set the transform pivot point to be the 3D Cursor, rather than the Median Point (which is the default) You should be able to scale without it changing the relations between objects. Generally best to use Ctrl-A to apply the scaling after as well.

No plans to put the addon up for sale, I'd like to always keep it free! (but I do accept donations on [Ko-Fi](https://ko-fi.com/measureit_arch) if you'd like to support development)

kevancress, 2021-05-15:
Hey @Blendersbrontobyte

I'm closing this issue for now, but let me know if you have any other issues or suggestions.

Thanks!

closed 2021-05-15 by kevancress