kevancress/MeasureIt_ARCH #174: Object line data isn't being referenced properly

SaitoPhoenix, 2021-04-29:
**Blender Version:** 2.90.1
**MeasureIt_ARCH Version:**  Development (0.5)

**Issue**
Line groups with the same name as an existing or previously existing line group share the vertex data for the existing line group.  The only exception to this is when creating a new line group where the previously existing line group was a "Line group by crease".  In this case the new vertex data appears to override the old.

The issue is pronounced by the fact that new line groups are named based off of list position.  Hence, if you create two line groups, they will be labeled "Line 1" and "Line 2", respectively.  If you delete the "Line 1" group, then the next line group you create will be "Line 2" again.  Since both are identically named, the newest "Line 2" group will use the earlier "Line 2" groups vertex data.

However, if you rename the newest "Line 2" group, it will switch back to the vertex data that was selected during its creation.

So it would seem that there reference bug that causes newly created line groups to use vertex data of previously created line groups with an identical name, instead of the vertex data for that particular line group instance.

**Expected Behavior**
1. Deleted line groups should have the references to their vertex data explicitly removed
2. Line groups should not be allowed to have the same name as another line group
3. Default line group names should never match an existing line group name.  If one exists, it should iterate to the next number in the sequence.

kevancress, 2021-05-07:
Hey @SaitoPhoenix

Thanks for the report,

Yeah this is an issue with how I'm caching Line Groups for drawing. Tabbing in and out of edit mode with the line group selected should refresh the cache and bring the new "Line 2" back to the data that was selected on creation. Refreshing the cache for all line groups when any are deleted should fix this, so I'll look into it this weekend. Having a properly unique name/ID like you suggested would help too. I'll let you know when I've got a working fix in the dev branch.

kevancress, 2021-08-30:
Added some fixes in 8e252aea36504745835c117ab770f24334d136ca that improve this a little, still not a total fix though.