kevancress/MeasureIt_ARCH #176: Cant Render Measurements from opposite direction

samucacsm, 2021-05-07:
[image]
Cant render measurements from a Ctrl-numpad 3 side view, only a numpad 3 side view. They appear normally on the viewport but when I hit the Render Image button they dont appear, only the contour lines.
I tried to delete the measurements and redo them, also tried to create a new camera, which worked on the numpad 3 side, but when moved to the other side, it simply wont render the measurements

kevancress, 2021-05-07:
Hey @samucacsm

There was a bug in the 0.4.6 release that was causing some issues with the raster image rendering. I've just put up a fix for it on the release page. Give it a shot and let me know if that fixes things.

It also might be an issue with the dimension placement system, setting the view plane to the YZ Plane in your dimension settings might fix it?

samucacsm, 2021-05-11:
thanks, I'm downloading it right now. 
One more thing, will line contours be available for instanced geometry nodes objects?

samucacsm, 2021-05-11:
> Hey @samucacsm
>
> There was a bug in the 0.4.6 release that was causing some issues with the raster image rendering. I've just put up a fix for it on the release page. Give it a shot and let me know [...]

So, I downloaded it, and it wont render the measurements if i select none. I gotta select YZ so it renders

[image]

kevancress, 2021-05-15:
Ahh okay, yeah I'm working on a total refactor of the dimension placement/drawing system that should fix this in future updates, but it may be a while before I get that working.

Also, I did a couple test with geometry nodes and it does seem like linework is being instanced properly so far!!

kevancress, 2022-07-13:
Took me a while but I think I've got this fixed in 7c6ee1b7c16f73f3e1b227673af894f99ff4df27

closed 2022-07-13 by kevancress