kevancress/MeasureIt_ARCH #177: Suggestions for better placement

mtonnie, 2021-05-23:
Anyone any suggestions how to improve placement for dimensions in the following example?

[image]

Thanks in advance

kevancress, 2021-05-23:
Hey @mtonnie 

Here's how I'd go about placing dimensions for the plan you've shown. A couple of notes,

- The Gizmo Button in the Measureit_ARCH main panel will show some gizmos below the dimension line in the 3D viewport, which lets you drag the dimension around to place it.
- Pressing "Tab" while dragging with the gizmo will flip the dimension (if you're using the latest development version, I just added that this week)
- You can also add dimensions to Blender's Empties, which is helpful for dimensioning to the midpoints of openings if necessary.

https://user-images.githubusercontent.com/16860574/119273347-767a7900-bbd8-11eb-8d66-f16ae8fda8fa.mp4

kevancress, 2021-05-23:
Closing this for now, but feel free to re-open if you have any other issues or suggestions!

closed 2021-05-23 by kevancress