kevancress/MeasureIt_ARCH #180: hatch only cross section

ekswathi, 2021-07-27:
Is there any way to hatch only the section area?

kevancress, 2021-08-30:
Hey @ekswathi

This is my usual setup for hatching cut area's:

[image]

I use a cube as a boolean object with a "cut" material that has a hatch set up on it. Then on the object I want to section I add the 'cut' material to an extra material slot that isn't applied to any faces, and a Boolean modifier.

When the object is sectioned, the faces cut by the Boolean modifier get the 'cut' material assigned (and the hatch too).

closed 2021-08-30 by kevancress

RadiumC, 2022-04-13:
Hello
When exporting as DXF, the lines are always yellow,continuous, and there are no hatches although its named after Line Style
Does it suppose to keep the line style from Measure Arch?