kevancress/MeasureIt_ARCH #181: Attempting to drag measurement offset gizmo produces a Python AttributeError

DaneHoffman, 2021-08-23:
When clicking the "Show Gizmos" button in the MeasureIt_ARCH Show/Hide panel, there is a small arrow that appears perpendicular to the measurement. Presumably this controls the offset of a dimension measurement, but when I click and drag it, it produces a Python error that says: 

`File "C:\path\to\addon\measureit_arch_dimensions.py", line 1568, in invoke dimension = getattr(myobj, self.dimType)`

Operating system is Windows 10, Blender version is 2.92.0

kevancress, 2021-08-29:
Hey @DaneHoffman are you using the latest development version from the main github page  there's some fixes to the gizmo system in there that might resolve this.

If its still an issue let me know!

kevancress, 2021-08-30:
Hey @DaneHoffman 

Just confirmed that this is working as expected in the latest development version.

[image]

Closing this for now but let me know if you find any other issues!

closed 2021-08-30 by kevancress