kevancress/MeasureIt_ARCH #182: how does the "Aligned" button work

rulon, 2021-08-29:
I tried to measure between 2 objects (cubes).
The tool shows a strange result.
What distance does it show here? 
Thanks.

[image]

kevancress, 2021-08-29:
Hey @rulon 

Yep that's definitely a bug. The aligned dimension in that case should show the straight line distance between the two object origins.

Try using the latest development version (if you're not already) that might fix it. You can download it from the main github page.

If it's  still a problem I'll take a look and see if I can reproduce and fix the issue tonight.

kevancress, 2021-08-30:
Hey @rulon 

Just confirmed that this is fixed and behaving as expected in the latest development version.

[image]

Closing this for now, but let me know if you run into any other issues!

closed 2021-08-30 by kevancress