kevancress/MeasureIt_ARCH #183: Plugin don't work under Blender 2.9 - Apple Silicon M1 Version

miarchimi, 2021-09-12:
I can install the plugin. But I think, its not full installed.
The Measurement-Settings under Screene-Properties are not visible.

If I click on Add Dimensions > Aligned (with selected edge), I get a error:

line 132, in __call__ ret = _op_cal(self.idname_py(), None, kw)
RuntimeError: Operator bpy.ops.object.shade_smooth.poll() failed, context is incorect

The plugin is not working

kevancress, 2021-09-13:
Hey @miarchimi,

Thanks for the report. I'll look into it this afternoon.

Two things to check:

1. I've found the most stable configuration is to be running the latest development version of MeasureIt_ARCH with Blender 2.93 or higher. There were some changes in blender API between 2.92 and 2.93 that may cause issues on versions 2.92 and lower

2. Based on the error message above, there could be a conflict with another add-on, try disabling your other addons, and then re-enabling MeasureIt_ARCH. That might fix the issue.

If either of those solves it let me know!
Thanks!

miarchimi, 2021-09-14:
Hey

I have installed the latest Silicon (M1) Blender version 2.93.4. All plugins are deactivated except for the standard and the plugin "Measurelt_ARCH 0.4.7" is installed. If I now want to activate the plugin, Blender immediately crashes. The plugin apparently causes problems with the Apple M1 version. You may also have an Apple M1 computer to test this out.

Greetings
Michel

miarchimi, 2021-09-17:
Here the generated Python error log after trying to activate the plugin. This log is from the lates developer version version, here form GitHub

# Python backtrace
  File "/Users/xy/Library/Application Support/Blender/2.93/scripts/addons/MeasureIt_ARCH-development/measureit_arch_geometry.py", line 93 in <module>
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 790 in exec_module
  File "<frozen importlib._bootstrap>", line 680 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 986 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1007 in _find_and_load
  File "/Users/xy/Library/Application Support/Blender/2.93/scripts/addons/MeasureIt_ARCH-development/measureit_arch_main.py", line 33 in <module>
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 790 in exec_module
  File "<frozen importlib._bootstrap>", line 680 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 986 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1007 in _find_and_load
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1058 in _handle_fromlist
  File "/Users/xy/Library/Application Support/Blender/2.93/scripts/addons/MeasureIt_ARCH-development/__init__.py", line 76 in <module>
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 790 in exec_module
  File "<frozen importlib._bootstrap>", line 680 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 986 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1007 in _find_and_load
  File "/Applications/Blender.app/Contents/Resources/2.93/scripts/modules/addon_utils.py", line 351 in enable
  File "/Applications/Blender.app/Contents/Resources/2.93/scripts/startup/bl_operators/userpref.py", line 439 in execute

kevancress, 2021-09-17:
Hey @miarchimi

Thanks for posting the backtrace, that's really helpful. Best I can tell the crash is happening when Blender is trying to register the custom point shader I'm using in the add-on. 

I'll do a quick once over of the shader code and see what might be causing the issue, I'd had some issues early on with the glsl shader code preforming oddly on Mac's and it makes sense that the M1 would have its own set of challenges in this area.

If you get a chance could you try running blender in gpu debug mode and try enabling the addon, this should give a more detailed log about what in the shader is causing the crash. 

you'll have to start blender from the terminal https://docs.blender.org/manual/en/latest/advanced/command_line/launch/macos.html 
with the argument "--debug-gpu"

Thanks!!!

miarchimi, 2021-09-18:
Hy Kevancress
First here the text output form the terminal and second the error log. I hope int helps you by finding the bug.
Cheers and have a nice weekend.
Michel

1)
Last login: Sat Sep 18 16:48:19 on ttys001
xy@xy ~ % Blender --debug-gpu
Read prefs: /Users/xy/Library/Application Support/Blender/2.93/config/userpref.blend
WARN (gpu.debug): source/blender/gpu/opengl/gl_debug.cc:177 init_gl_callbacks: Failed to hook OpenGL debug callback. Use fallback debug layer.
WARN (gpu.debug): source/blender/gpu/opengl/gl_debug.cc:177 init_gl_callbacks: Failed to hook OpenGL debug callback. Use fallback debug layer.
WARN (gpu.shader): OVERLAY_shader_armature_sphere_410 FragShader: 
      | 
      | Warning: extension 'GL_ARB_conservative_depth' is not supported

WARN (gpu.shader): OVERLAY_shader_armature_sphere_397 FragShader: 
      | 
      | Warning: extension 'GL_ARB_conservative_depth' is not supported

WARN (gpu.debug): source/blender/gpu/opengl/gl_debug.cc:177 init_gl_callbacks: Failed to hook OpenGL debug callback. Use fallback debug layer.
WARN (gpu.debug): source/blender/gpu/opengl/gl_debug.cc:177 init_gl_callbacks: Failed to hook OpenGL debug callback. Use fallback debug layer.
addon_utils.disable: MeasureIt_ARCH-development not disabled
Modules Installed (MeasureIt_ARCH-development) from '/Users/xy/Desktop/MeasureIt_ARCH-development.zip' into '/Users/xy/Library/Application Support/Blender/2.93/scripts/addons'
M_ARCH import modules
Writing: /var/folders/q4/36r5ybsj6jbgnm889l_sjq200000gq/T/blender.crash.txt
zsh: segmentation fault  Blender --debug-gpu
xy@xy ~ % 

2)
# Blender 2.93.4, Commit date: 2021-08-31 09:23, Hash xy
bpy.data.window_managers["WinMan"].addon_search = "mea"  # Property
bpy.context.space_data.system_bookmarks_active = 1  # Property
Modules Installed (MeasureIt_ARCH-development) from '/Users/xy/Desktop/MeasureIt_ARCH-development.zip' into '/Users/xy/Library/Application Support/Blender/2.93/scripts/addons'  # Info

# backtrace
0   Blender                             0x000000010ba47ec4 BLI_system_backtrace + 56
1   Blender                             0x0000000104e08178 sig_handle_crash + 364
2   libsystem_platform.dylib            0x000000019430ec44 _sigtramp + 56
3   libGLProgrammability.dylib          0x00000001d649e6dc glpLLVMGetFunctionGlobalVariableUse + 112
4   libGLProgrammability.dylib          0x00000001d649e918 glpLLVMGetFunctionGlobalVariableUse + 684
5   libGLProgrammability.dylib          0x00000001d649e918 glpLLVMGetFunctionGlobalVariableUse + 684
6   libGLProgrammability.dylib          0x00000001d649e918 glpLLVMGetFunctionGlobalVariableUse + 684
7   libGLProgrammability.dylib          0x00000001d649e918 glpLLVMGetFunctionGlobalVariableUse + 684
8   libGLProgrammability.dylib          0x00000001d6499698 glpLLVMCGFindSamplersAndBuffers + 1380
9   libGLProgrammability.dylib          0x00000001d6498544 glpLLVMCGTopLevel + 2900
10  libGLProgrammability.dylib          0x00000001d64bf924 glpLinkProgram + 10388
11  libGLProgrammability.dylib          0x00000001d64d9598 ShLink + 208
12  GLEngine                            0x00000001d6694a00 gleLinkProgram + 1212
13  GLEngine                            0x00000001d6603cac glLinkProgramARB_Exec + 256
14  Blender                             0x0000000109f21f64 _ZN7blender3gpu5debugL19debug_glLinkProgramEj + 44
15  Blender                             0x0000000109f24d04 _ZN7blender3gpu8GLShader8finalizeEv + 96
16  Blender                             0x0000000109f15994 GPU_shader_create_ex + 880
17  Blender                             0x0000000109f15fe8 GPU_shader_create_from_python + 200
18  Blender                             0x0000000106043c74 pygpu_shader__tp_new + 116
19  Blender                             0x000000010a071aa4 type_call + 84
20  Blender                             0x000000010a019084 _PyObject_MakeTpCall + 360
21  Blender                             0x000000010a0eb25c call_function + 512
22  Blender                             0x000000010a0e8c6c _PyEval_EvalFrameDefault + 23572
23  Blender                             0x000000010a0ec14c _PyEval_EvalCode + 3032
24  Blender                             0x000000010a0e2f90 PyEval_EvalCode + 80
25  Blender                             0x000000010a0dfde8 builtin_exec + 648
26  Blender                             0x000000010a059d8c cfunction_vectorcall_FASTCALL + 248
27  Blender                             0x000000010a0e8e6c _PyEval_EvalFrameDefault + 24084
28  Blender                             0x000000010a0ec14c _PyEval_EvalCode + 3032
29  Blender                             0x000000010a019800 _PyFunction_Vectorcall + 256
30  Blender                             0x000000010a0eb1fc call_function + 416
31  Blender                             0x000000010a0e8b74 _PyEval_EvalFrameDefault + 23324
32  Blender                             0x000000010a01987c function_code_fastcall + 112
33  Blender                             0x000000010a0eb1fc call_function + 416
34  Blender                             0x000000010a0e8b50 _PyEval_EvalFrameDefault + 23288
[... 64 lines cut ...]
99  Blender                             0x000000010a0e8bf0 _PyEval_EvalFrameDefault + 23448

# Python backtrace
  File "/Users/xy/Library/Application Support/Blender/2.93/scripts/addons/MeasureIt_ARCH-development/measureit_arch_geometry.py", line 93 in <module>
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 790 in exec_module
[... cut ...]
  File "<frozen importlib._bootstrap_external>", line 790 in exec_module
  File "<frozen importlib._bootstrap>", line 680 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 986 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1007 in _find_and_load
  File "/Applications/Blender.app/Contents/Resources/2.93/scripts/modules/addon_utils.py", line 351 in enable
  File "/Applications/Blender.app/Contents/Resources/2.93/scripts/startup/bl_operators/userpref.py", line 439 in execute

miarchimi, 2021-09-18:
I have an idea for the next version you could integrate at a future Version. A dynamic plan border for different paper formats as well as an orthographic camera with automatic correct settings to render a plan in correctly scale dimensions, e.g. 1:50, etc., and then this image could be printed true to scale.

kevancress, 2021-09-18:
@miarchimi 

Thanks for the logs that helps a lot! Looks like a had a couple of places in my shader code that were implicitly casting int's to floats, which I think was the source of the problem! can you give the latest development version a try and let me know if it works?

I like the Dynamic plan border idea, sounds great for making title blocks that adjust to fit a given paper size. The view settings in MeasureIt_ARCH do already have a paper size & scale setting for setting up true to scale drawing though!

miarchimi, 2021-09-19:
Sadly it still does not work ...

Last login: Sun Sep 19 19:11:01 on ttys000
xy@xy ~ % Blender --debug-gpu
Read prefs: /Users/xy/Library/Application Support/Blender/2.93/config/userpref.blend
WARN (gpu.debug): source/blender/gpu/opengl/gl_debug.cc:177 init_gl_callbacks: Failed to hook OpenGL debug callback. Use fallback debug layer.
WARN (gpu.debug): source/blender/gpu/opengl/gl_debug.cc:177 init_gl_callbacks: Failed to hook OpenGL debug callback. Use fallback debug layer.
WARN (gpu.shader): OVERLAY_shader_armature_sphere_410 FragShader: 
      | 
      | Warning: extension 'GL_ARB_conservative_depth' is not supported

WARN (gpu.shader): OVERLAY_shader_armature_sphere_397 FragShader: 
      | 
      | Warning: extension 'GL_ARB_conservative_depth' is not supported

WARN (gpu.debug): source/blender/gpu/opengl/gl_debug.cc:177 init_gl_callbacks: Failed to hook OpenGL debug callback. Use fallback debug layer.
WARN (gpu.debug): source/blender/gpu/opengl/gl_debug.cc:177 init_gl_callbacks: Failed to hook OpenGL debug callback. Use fallback debug layer.
addon_utils.disable: MeasureIt_ARCH-development not disabled
Modules Installed (MeasureIt_ARCH-development) from '/Users/xy/Desktop/MeasureIt_ARCH-development.zip' into '/Users/xy/Library/Application Support/Blender/2.93/scripts/addons'
M_ARCH import modules
Writing: /var/folders/q4/36r5ybsj6jbgnm889l_sjq200000gq/T/blender.crash.txt
zsh: segmentation fault  Blender --debug-gpu
xy@xy ~ % Blender --debug-gpu
Read prefs: /Users/xy/Library/Application Support/Blender/2.93/config/userpref.blend
WARN (gpu.debug): source/blender/gpu/opengl/gl_debug.cc:177 init_gl_callbacks: Failed to hook OpenGL debug callback. Use fallback debug layer.
WARN (gpu.debug): source/blender/gpu/opengl/gl_debug.cc:177 init_gl_callbacks: Failed to hook OpenGL debug callback. Use fallback debug layer.
WARN (gpu.shader): OVERLAY_shader_armature_sphere_410 FragShader: 
      | 
      | Warning: extension 'GL_ARB_conservative_depth' is not supported

WARN (gpu.shader): OVERLAY_shader_armature_sphere_397 FragShader: 
      | 
      | Warning: extension 'GL_ARB_conservative_depth' is not supported

WARN (gpu.debug): source/blender/gpu/opengl/gl_debug.cc:177 init_gl_callbacks: Failed to hook OpenGL debug callback. Use fallback debug layer.
M_ARCH import modules
Writing: /var/folders/q4/36r5ybsj6jbgnm889l_sjq200000gq/T/blender.crash.txt
zsh: segmentation fault  Blender --debug-gpu
xy@xy ~ % 

[image]

# Python backtrace
  File "/Users/xy/Library/Application Support/Blender/2.93/scripts/addons/MeasureIt_ARCH-development/measureit_arch_geometry.py", line 93 in <module>
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 790 in exec_module
  File "<frozen importlib._bootstrap>", line 680 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 986 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1007 in _find_and_load
  File "/Users/xy/Library/Application Support/Blender/2.93/scripts/addons/MeasureIt_ARCH-development/measureit_arch_main.py", line 33 in <module>
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 790 in exec_module
  File "<frozen importlib._bootstrap>", line 680 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 986 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1007 in _find_and_load
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1058 in _handle_fromlist
  File "/Users/xy/Library/Application Support/Blender/2.93/scripts/addons/MeasureIt_ARCH-development/__init__.py", line 76 in <module>
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 790 in exec_module
  File "<frozen importlib._bootstrap>", line 680 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 986 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1007 in _find_and_load
  File "/Applications/Blender.app/Contents/Resources/2.93/scripts/modules/addon_utils.py", line 351 in enable
  File "/Applications/Blender.app/Contents/Resources/2.93/scripts/startup/bl_operators/userpref.py", line 439 in execute

kevancress, 2021-09-20:
Ahh darn!

Looks like it's still crashing at the point shader. I did some more cleanup to the shader code, removing some unnecessary global variables, hopefully that fixes it!

If you could download the latest development version and give it a test that would be amazing, much appreciated!

(I should really see if I can get an M1 Mac so I can test on my end)

miarchimi, 2021-09-20:
Yeah ... 👍 now it works. Great job, thanks a lot!
Best Greetings
Michel

[image]

miarchimi, 2021-09-20:
I test now a little bit the Tool. I think it works stable.

Some questions I have:

1) How can I set dimensions easy inside or outside of a e.g. Wall at creation-time?
For example with Checkboxes named like (top, bottom, left, right)
On the other hand I have to modify each dimension afterward over the "tweak-distance-field" afterwards ...

2) How can I rotate dashes 90°?

[image]

Thanks so log that I generally works.

Greetings Michel

kevancress, 2021-09-27:
Hey @miarchimi 

1) Pressing Tab while selecting the tweak distance gizmo will flip the dimension to the outside.

[image]

I like the idea of the top/bottom left/right checkboxes though. right now the dimension placement always defaults to pointing away from the surface normals of the edge its attached to, (or defaults to top and left if there are no surfaces)

2) No way to rotate the dashes currently. I'll look into adding that

miarchimi, 2021-09-28:
I tested the plugin a little more. I inserted dimension lines and deleted them again. Then the selection was by a empty field from which I wanted to delete. Then an error appeared. Now I have two dimensions in the 3D window, but they no longer appear in the list. So now I have noch chance to delete this measurements any more. Is there a workaround to delete such measurements on an other way? A button that cleans up the arrays would also be good if such unexpected errors occur.

[image]

kevancress, 2021-09-28:
Hey @miarchimi 

There is an option to delete all if this happens:
[image]
 

It's odd that the dimensions weren't deleted properly the first time, were any error reported when initially deleting the dimensions from the list?

miarchimi, 2021-09-29:
The error is also displayed if the list empty and you clicked on delete.