kevancress/MeasureIt_ARCH #185: line groups/line style/annotation text update/suggestion

mizo1900, 2021-09-24:
**line group delete**
when a line group " say line group 1 " is assigned to a number of vertices to outline a specific edge path , then delete" line group 1 " from line groups to add a new vertices to the selection , the new created line group takes the previous name line group 1 and discard the new selected vertices , so i have to change the line group name to something else to have the new selected vertices add to that new line group

**line style rename**
when a line style is assigned to line groups , the style name has to stay the same, if renamed to something else i have to reassign the new style name to that line group

**annotation update**
when type annotation txt it does not update in viewport unless i click on that text field

**line groups selection suggestion :**
due line groups complexity when have a long list of line groups, is not possible to remember which line group is assigned to a specific group of edges , so it would be great to add a feature when a group of edges are selected in the viewport it will also highlight the line groups which those edges belong to

kevancress, 2021-09-27:
Hey @mizo1900 

These are great suggestions. I just pushed a fix that should resolve the Line Group Delete issue you mentioned (13e1b37f1051f64f3d369f9160e93e72d90eaa41). I'll do some work on the others this week too.

kevancress, 2021-09-27:
@mizo1900 

Just added the Line Group Selection change too! (64ec92a20ae0dd2c6eb24b6e4ae57f39a1eaa65c):

[image]

I've done it so that the active line group is show in its defined color, while the others on the selected object are highlighted in blenders selection color, normally I'd do it the other way around, but that makes it a bit of a pain to edit the colors.

The other two suggestions may be a bit more of a challenge. 

The style renaming issue has been bothering me too for a while and I haven't been able to find a fix, but I'll give it another look!

The annotation update is more of a performance limitation. For the annotation to update in the 3D view I have to render that annotation text to a texture offscreen and then pass it back to the annotation object, which then passes it to the gpu to draw when the viewport updates... so I've limited it to only update when the text field or color changes... updating every frame would be really slow, and I'm not sure we could update the annotation as you type, since I think the value of the text field only gets updated (as far as blender scripts are aware) when you finish typing and leave the textbox...

kevancress, 2021-09-28:
@mizo1900 

This is a bit of a hack, but I think I've got style names updating. On objects at least. the default styles in the ui for new dimensions still have to be manually re-assigned.

[image]

I just added an option for styles to keep track of their previous name. Then when ever a dimension (or annotation or line group) gets drawn, if it cant find it's style it checks for the previous names too, and if it finds a match there it updates itself to use the new name. ( Commit is here: 39d18a9800d0f736a3d6014779a426625adb1ef1)

Not the best fix in the world but, better than having to manually re-assign a bunch of styles all over the place!

kevancress, 2021-09-28:
Going to close this for now and open a separate thread for the Annotation update while typing request... again I'm not sure if that one will be possible but I'll keep an issue for it as a reminder

closed 2021-09-28 by kevancress