kevancress/MeasureIt_ARCH #198: error message when deleting any object from the scene

mizo1900, 2021-11-23:
[image]
Traceback (most recent call last):
  File "C:\Users\DELL\AppData\Roaming\Blender Foundation\Blender\2.83\scripts\addons\MeasureIt_ARCH\measureit_arch_main.py", line 306, in draw
    col = layout.column(align=True, heading='Settings')
TypeError: UILayout.column(): takes at most 1 arguments, got 2

location: <unknown location>:-1

RadiumC, 2022-04-10:
I confirm the same issue

kevancress, 2022-04-12:
@mizo1900 @RadiumC  

I haven't been able to reproduce this error in the latest development version.

could you post more detailed steps, or a demo file?

kevancress, 2022-07-13:
Closing for now, let me know if this is still an issue!

closed 2022-07-13 by kevancress