kevancress/MeasureIt_ARCH #201: Dimension and Annotation Text disappears when changing font

johannes-wilde, 2022-01-24:
I can't really reproduce the bug on purpose. With a clean startup file it works for a while, the more complex it gets, the bug appears sooner or later. Both Blender 3.0 / 3.1
PopOS 21/10

kevancress, 2022-01-25:
Hey @johannes-wilde,

This likely has something to do with how I'm handling the text drawing. The text itself gets drawn to an offscreen texture buffer and then rendered onto a card  attached to the dimension. There's a bunch of places in this process where issues can arise.

A couple questions:

Does the text re-appear after you resize or move the mesh or does it stay missing?

Also are there any particular fonts that this regularly happens with?

Is it just the text that disappears or the entire dimension?

Are you changing the font through a style or directly on the dimension / annotation itself?

I'll try to look into this over the next couple days!

kevancress, 2022-01-25:
@johannes-wilde 

Ahh I think I've managed to reproduce the issue!

This is the same as what you're experiencing?
[image]

kevancress, 2022-01-25:
Okay so I think this has to do with a change to how the default font gets assigned in Blender 3.0+

I've commited a fix here https://github.com/kevancress/MeasureIt_ARCH/commit/2f5c42c1364acc6625f458ddbf00d0893f452a60

That seems to fix the issue on windows at least. Give the latest development version a try and let me know if that fixes it for you!

johannes-wilde, 2022-01-25:
Hey!
That's just the behavior that I experienced. Your fix seems to work on a fresh file for me - I played around for some minutes with different meshes and for now it works fine.
However I have file, where I used MeasureIt Arch before, that still shows the bug. Is there some data cached from a previous version of the addon?
I just attach the simplified version
[klausenerstr_39_006.zip](https://github.com/kevancress/MeasureIt_ARCH/files/7931537/klausenerstr_39_006.zip)
.

johannes-wilde, 2022-01-25:
Small update:
I appended the mesh in a fresh file and that works.

kevancress, 2022-01-25:
Hmm, must be more than one issue causing this.

Thanks for attaching the file, that really helps! I'll take a look and see what I can find

johannes-wilde, 2022-01-26:
Here if have the case that the first dimensions work as expected, while the later added dimensions are not rendered.

[image]