kevancress/MeasureIt_ARCH #202: Cannot load the addon

c1symeon, 2022-02-14:
Blender 3.0 under Windows 10
I cannot load the addon.  I get this error message:
`Traceback (most recent call last):
  File "C:\Program Files\Blender Foundation\Blender 3.0\3.0\scripts\modules\addon_utils.py", line 351, in enable
    mod = __import__(module_name)
ModuleNotFoundError: No module named 'MeasureIt_ARCH-mARCHv0'
`
I have tried installing both in the Appdata subfolder and in the ProgramFiles subfolder and get the same message.

kevancress, 2022-04-03:
Hey @c1symeon sorry for the slow reply.

The 0.4.6 version is quite a bit behind, since I haven't put together a proper release in a while. The development branch of the addon should work with 3.0 +,  I've been using it with Blender 3.1 myself

kevancress, 2022-04-03:
Updated the 0.4.6 release page to indicate that is doesn't work with 3.0 + and that the development branch should be used instead.

closed 2022-04-03 by kevancress