kevancress/MeasureIt_ARCH #203: Is this addon being updated to Blender 3.0?

baldyman01, 2022-02-14:
Is this addon being updated to Blender 3.0? There a re numerous features that throw tracebacks when used in 3.0, which I'm assuming is because I shouldn't be using it on 3.0 in the first place so my bad.
I'd like to see this updated though if you are stlll working on it.

KyleTryon, 2022-03-17:
Came here looking for the same thing. I noticed the last release was a year ago, so I am wondering now if the project is still active.

This commit seems to suggest maybe 3.1 is being worked on somewhat recently: https://github.com/kevancress/MeasureIt_ARCH/commit/9e0ce66532af77e63492e08ec6640e4142fd0e70

kevancress, 2022-04-03:
Hey @baldyman01 @KyleTryon 

I haven't had much time for development the last little while, so haven't been putting together proper releases lately, but the development branch should work with Blender 3.0+. I've been using it with 3.1 myself mostly.

baldyman01, 2022-04-14:
Sorry for the delay, family stuff going on. I'm running on 3.1.2 now and so far it seems to be behaving. I noticed that I didn't get my 3.1 settings imported when I started using 3.1.2, so I decided to make a clean break and started just adding things back in as I go.  So far so good. 

Consider this fixed. If anything crops up I'll come back.

Cheers

Pete

closed 2022-04-16 by kevancress