kevancress/MeasureIt_ARCH #209: Error when trying to use Measure it Arch Render

RadiumC, 2022-04-10:
I cannot render with the Measure It Arch render feature
When tring any of the three buttons, it throws error 
PermissionError:[Win Error5] Access is denied `Renders
[image]
`

kevancress, 2022-04-12:
Hey @RadiumC

This Error occurs when the output path defined in the active view either doesn't exist, or is a folder you don't have permission to access.

I should add a more descriptive error message, and an option to create the

RadiumC, 2022-04-12:
Thanks, it works now
How do I make it work with Blender Bim? Do i just need to change the name of the folder?
Where do I get the latest git branch?

RadiumC, 2022-04-12:
I think the description is incomplete in the Readme file?

Latest git master
[Install Blender 2.8 or higher](https://www.blender.org/download/)
Locate your platform and Blender installation specific Blender addons directory:
On Windows, this is usually /Program Files/Blender Foundation/Blender BLENDER_VERSION/BLENDER_VERSION/scripts/addons
On Mac, this is usually /Applications/Blender.app/Contents/Resources/BLENDER_VERSION/scripts/addons

kevancress, 2022-04-12:
@RadiumC 

Sorry yeah the docs are a bit out of date.

Changing the output path to an existing folder should work, and renders should automatically be saved there after you hit the render button.

to get the latest version of the addon go to the main git page, and go to the "development" branch. Download that and extract the addon into the Blender Addons Directory (the paths you mentioned above) and then it should work. You can also use the install button from the Blender Addons Menu.

I'd recommend using the Addon with Blender 3.1 Stable. That's what I'm working in these days, so it's most stable there.

I haven't done much work using both MeasureIt_ARCH and Blender BIM. We did some integration tests about a year ago, but I haven't had time to keep up with the Blender BIM team (but they're doing amazing stuff!). Most of my work is single family residential too, so I'm generally not needing a full BIM workflow in my day-to-day.

RadiumC, 2022-04-13:
Thanks for replying, 
I think i have the development branch because the folder is named  MeasureIt_ARCH-development
Do i need to rename it and change the dash "-" into underscore "_" to get recognized by Blender Bim? 
Using 3.1 Stable here too as well.

kevancress, 2022-04-16:
Hey @Radium 

Yeah my approach to plans and sections is generally to have all the 3D elements in my scene set up with a Boolean difference modifier all using a single large cube as the cutting object. Objects cut with a Boolean modifier will actually take on the material of the cutting object, as long as the source object has a material slot with the cut material available, which is how I do a simple hatch on cut faces. Then I just set up my views on different frames, and move and keyframe the cutting object depending on if its a plan, section etc. 

Kinda hard to describe but here's a video showing my usual set up on a demo project:

https://user-images.githubusercontent.com/16860574/163683707-2ce216d6-51dd-4c7e-a8f8-b72c0f434c1a.mp4

I usually add a bit of detailing just in 2D for each particular view as well, makes it pretty quick to pull together concept plans:

[image]

kevancress, 2022-04-16:
I'll have to look into the BlenderBIM integration stuff later when I get some more time. It's been a while since I've looked at that code and what it takes to get them working together

RadiumC, 2022-04-18:
Thank you for the tips and the video !
Did you import the symbols for doors/windows from standard DWG libraries?
By the way, is the Hatch feature part of your addon or is it in the standard Blender? Or is it from Blender Bim? I only recently noticed it and its really powerfull!

kevancress, 2022-04-18:
@RadiumC No problem!

For 2D symbols I either make them in Blender, or use the built in .dxf importer to bring in blocks exported from a .dwg

The hatch feature panel in the material properties is from MeasureIt_ARCH yeah, I think blenderBIM has a way of doing hatches too though.

RadiumC, 2022-04-18:
@kevancress Awesome feature! How do I use the Pattern setting for the Hatch? I see there is a collection picke but  what does it need to have for a pattern?

kevancress, 2022-04-19:
@RadiumC  If you've got a collection selected then the hatch will draw all the edges in that collection as the hatch pattern.

It tiles based on the 0m-1m square in the xy plane. So I usually set up a separate scene and make my hatch collections there:

[image]

RadiumC, 2022-04-20:
Hmm i wasn`t able to generate any hatch. Maybe there was something with the thickness or something else.  I did create a series of edges at  world center from 0 to 1m.
Do you mind if you can show an example of setting up a hatch?

closed 2022-06-17 by RadiumC