kevancress/MeasureIt_ARCH #213: MeasureIt_ARCH Vector option throwing traceback

baldyman01, 2022-04-14:
MeasureIt_ARCH 1.8.1 On Blender 3.1.2 and Ubuntu 21.04

Rendering an image works as expected, but I'm trying to use the Vector option to render an svg. I have a default cube with a single Aligned dimension on it. Clicking the Vector button produces the following traceback - 

Python: Traceback (most recent call last):
  File "/home/pete/.config/blender/3.1/scripts/addons/MeasureIt_ARCH/measureit_arch_render.py", line 232, in execute
    if render_main_svg(self, context):
  File "/home/pete/.config/blender/3.1/scripts/addons/MeasureIt_ARCH/measureit_arch_render.py", line 517, in render_main_svg
    draw3d_loop(context, objlist, svg=svg)
  File "/home/pete/.config/blender/3.1/scripts/addons/MeasureIt_ARCH/measureit_arch_geometry.py", line 3742, in draw3d_loop
    draw_hatches(context, myobj, scene.HatchGenerator, mat, svg=svg)
  File "/home/pete/.config/blender/3.1/scripts/addons/MeasureIt_ARCH/measureit_arch_geometry.py", line 350, in draw_hatches
    bm.from_object(myobj, bpy.context.view_layer.depsgraph, deform=True)
SystemError: More keyword list entries (5) than format specifiers (4)

location: <unknown location>:-1

Anything you want me to try, let me know.

kevancress, 2022-04-16:
@baldyman01 Yeah this is an issue due to some changes in the api for the bm.from_object() call. This makes MeasureIt_ARCH 0.4.6 incompatible with Blender 3.0+.

This is fixed in the latest development branch though, I'd recommend using the development branch in general, but especially if you're using any blender version later than 3.0.

baldyman01, 2022-04-16:
Ah, OK was unaware of the availability of that. Happy to give it spin. 

Cheers Kevan!

kevancress, 2022-04-29:
@baldyman01 Closing this one for now, but let me know if there's any other issues

closed 2022-04-29 by kevancress