kevancress/MeasureIt_ARCH #219: Measurements missing and different direction than viewport

BaxterBlender, 2022-06-06:
Hi! I am having some problems. In my viewport I see all dimensions, in render only 2. 
I'm not sure what could be wrong, very simple scene, the elements on the left are correct, the ones on the right should be too, did the same way, but only show partially.
[image]

At the bottom we can see measureit_arch output and it is completely different than the preview. Sorry, I'm new here and I have no idea how to attach blend file. GitHub doesn't accept them as valid attachments.

kevancress, 2022-07-05:
Hey @BaxterBlender 

Sorry for the slow reply, think I managed to fix this (9ff04e476d4ba6a419c33c2fd90f856a0b6e22a0 , 7c6ee1b7c16f73f3e1b227673af894f99ff4df27) 

Try out the latest version in the development branch and let me know if it works.

The placement is still a little off in some situations, but it should be consistent between the viewport and render at least

BaxterBlender, 2022-07-05:
Hi! 
Thanks a lot! I'm delighted to see that you are constantly improving. 
Without addons, like yours, I wouldn't be able to use Blender for technical drawings. 
Keep up the great work you're doing! 

⁣Sent from Blue ​

***@***.***>

BaxterBlender, 2022-07-20:
Hi Kevan!

Unfortunately one small bug. I tried on very simple objects AREA
measurement. And... the measurements don't show up, and when I try to
render them, I get this:

Python: Traceback (most recent call last):
  File "C:\Users\Baxter\AppData\Roaming\Blender
Foundation\Blender\3.0\scripts\addons\MeasureIt_ARCH-development\measureit_arch_render.py",
line 107, in execute
    outpath = render_main(self, context)
  File "C:\Users\Baxter\AppData\Roaming\Blender
Foundation\Blender\3.0\scripts\addons\MeasureIt_ARCH-development\measureit_arch_render.py",
line 252, in render_main
    draw3d_loop(context, objlist)
  File "C:\Users\Baxter\AppData\Roaming\Blender
Foundation\Blender\3.0\scripts\addons\MeasureIt_ARCH-development\measureit_arch_geometry.py",
line 3613, in draw3d_loop
    draw_areaDimension(context, myobj, DimGen,
  File "C:\Users\Baxter\AppData\Roaming\Blender
Foundation\Blender\3.0\scripts\addons\MeasureIt_ARCH-development\measureit_arch_geometry.py",
line 1738, in draw_areaDimension
    placementResults = setup_dim_text(myobj,dim,dimProps,
sumArea,origin,vecX,vecY, is_area=True)
  File "C:\Users\Baxter\AppData\Roaming\Blender
Foundation\Blender\3.0\scripts\addons\MeasureIt_ARCH-development\measureit_arch_geometry.py",
line 3692, in setup_dim_text
    distanceText = format_area(dist, dim=dim)
TypeError: format_area() got an unexpected keyword argument 'dim'

location: <unknown location>:-1

I'm attaching my blend file.
If there is anything in the future I can do to help you, let me know!

Best Regards
Baxter

BaxterBlender, 2022-07-29:
Dear Kevan.

As always, congrats on your great work.
I have a question. Should I report to you when I find something not
working, or you have so much to do I should let it go and wait for your
solution as you have so much work, adding even more will bring no results?
I don't want to be perceived as a problematic guy :)

I found one more.
If I add a one-axis dimension, all is fine. As soon as I add another, they
become "linked" and both refer to the first one, without possibility to do
anything with the second one. In viewport I see both, but controls for both
change only the first one. I am attaching a short video showing what I
mean.

Available here:

https://drive.google.com/file/d/1Ajsh849-SKw__7gZD9HgqzdMLPvMJCMH/view?usp=sharing