kevancress/MeasureIt_ARCH #220: Render Path Not Compatible with Linux

BerlinAW, 2022-06-08:
Hello, 

There is a slight problem with the Measure It Arch plugin, on other OS's that leads to a little bit of an interoperability issue between windows, linux and file sharing protocols.

Issue:
When I click the "MeasureIt_ARCH Image" button on Linux (Manjaro running the 5.15.32-1 Kernel) the plugin creates a folder with named "Renders\" This does create a bit of a problem on linux as that is the escape character. This also makes it impossible to sync the directory with nextcloud or a samba share, without renaming the directory. This problem naturally does not appear on Windows.

In general though, your plugin is fantastic and has virtually removed the need for having sketchup and sketchup layout in my workflow. Thanks for all the work on this plugin. Please let me know if there is anything else that I need to clarify.

BerlinAW, 2022-06-08:
I should say that I am using Version 0.5.0

kevancress, 2022-07-12:
Hey @BerlinAW

Sorry for the slow response! Going to try to look into this one today. Not sure I entirely understand the issue though (Windows user who's only dabbled a little in Linux). Which character in the file path in particular is causing the issue? Could you provide an example of a file path the would be valid in the Linux file system?

I've got a new partition with PopOS set up on it now too so I can do some testing myself as well

Thanks!

BerlinAW, 2022-07-12:
Hello @kevancress . 

Not a problem. Let me see if I can explain the situation a little bit better.

The crux of the problem is on Linux operating systems the opposite slash is used to indicate directories or a path, in comparison to windows.

----Example Folder Path----
Linux:  /mnt/Storage/StorageHDD/MeasureIt_Arch_Example/
Windows: C:\Users\Administrator\MeasureIt_Arch_Example\

So when the MeasureIt_Arch creates the line drawing render, the plugin first creates the Renders folder. Which is really cool in and of itself. The problem is that for some reason, the plugin creates "Renders\" instead of just "Renders".

Linux allows the folder to exist, which is also kind of interesting, and the plugin seems to be able to use it. The problem shows up if I try to copy and paste or sync that directory with another server or computer. The folder can't be moved because of that extra "\" character. Backslash on linux systems is the escape character. So if I was typing along in command line and I wanted the bash shell to ignore a particular character, I would include a "\" in front of it. I am certain that is the thing that is causing the issue. 

Sidebar: I have heard good things about PopOS. I have never played with it myself. But it seems to be the linux distro of choice for developers. I think HP is shipping it on one of their shiny new laptops. I can also recommend Fedora and Manjaro. 

Lemme know if there is anything else I can do to help clarify.

kevancress, 2022-07-13:
@BerlinAW 

Okay so I haven't tested on Linux yet, but I think this may have been as simple as adjusting the default output path in the view properties, give the latest development branch a shot and let me know if it works! I'll test myself too later tonight.

BerlinAW, 2022-07-15:
Fantastic. Single Frame, Animation and SVG all seem to work well. 

So stupid question, did I miss a user editable output path where I could have changed that?

Regardless, thanks for looking into this.

BerlinAW, 2022-07-15:
@kevancress 

Okay based on your comments on the python files in GitHub, I started asking myself, "If this is user editable, why I didn't see this before?" It turns out that in the properties window > scene properties > MeasureIt_Arch > Views > View # settings, is completely blank. _I don't normally interact with views since most of my renders are just single angles._  If I create a new view, the settings are still blank. This is for Blender 3.2.0.

[image]

This actually might be related to another little bug that I ran across. Sometimes if I duplicate an object with measurements. Not all of the measurements show up in the object's dimension list, but are still visible in the 3D view port. The only way to clear these ghost dimensions is to use the "Delete All Dimensions" option from the drop down. 

I realize this is all not directly related to the original problem. Would you like me to open up a new ticket?

Thanks.

BerlinAW, 2022-07-15:
You can scratch that. I figured it out. I need to assign a camera to the view, then I can see the output options. 

I wonder if that might not be another good default? Automatically assign the default camera to view 1?

kevancress, 2022-08-28:
Hey @BerlinAW

Sorry it took me a while to circle back to these! I've opened up a new issue for the problem of dimensions not showing up in the UI list when an object is duplicated!

Also, yeah I totally agree that having the view settings not show up until a camera is set makes them really easy to miss! I've changed how this shows up now so that the settings are still visible, just disabled until a camera is set. 

I'm hesitant to automatically assign a camera, since it might cause bugs for people with custom blender startup scenes...

Anyway going to close this issue for now, but let me know if you come across any other problems!

Thanks,
Kevan

closed 2022-08-28 by kevancress