kevancress/MeasureIt_ARCH #224: Area not displayed + Measureit_ARCH Image traceback

baldyman01, 2022-07-07:
I downloaded the development code today 7/7. Area's now can be added and deleted but never display in viewport. Trying to Render a MeasureitARCH  image throws the following error -

Python: Traceback (most recent call last):
  File "/home/pete/.config/blender/3.2/scripts/addons/MeasureIt_ARCH-development/measureit_arch_render.py", line 107, in execute
    outpath = render_main(self, context)
  File "/home/pete/.config/blender/3.2/scripts/addons/MeasureIt_ARCH-development/measureit_arch_render.py", line 252, in render_main
    draw3d_loop(context, objlist)
  File "/home/pete/.config/blender/3.2/scripts/addons/MeasureIt_ARCH-development/measureit_arch_geometry.py", line 3613, in draw3d_loop
    draw_areaDimension(context, myobj, DimGen,
  File "/home/pete/.config/blender/3.2/scripts/addons/MeasureIt_ARCH-development/measureit_arch_geometry.py", line 1738, in draw_areaDimension
    placementResults = setup_dim_text(myobj,dim,dimProps, sumArea,origin,vecX,vecY, is_area=True)
  File "/home/pete/.config/blender/3.2/scripts/addons/MeasureIt_ARCH-development/measureit_arch_geometry.py", line 3692, in setup_dim_text
    distanceText = format_area(dist, dim=dim)
TypeError: format_area() got an unexpected keyword argument 'dim'

Blender 3.2.0 Stable
Measureit_ARCH development as of 7/7

baldyman01, 2022-07-12:
Yep, this issue is fixed. Nice job!

baldyman01, 2022-07-12:
I also note that I'm not corrupting existing dimensions by editing the owner object any more. Not sure if it was a deliberate fix, but it's a lot better.

kevancress, 2022-07-12:
Great! hmm nope I haven't done anything intentional to fix the dims getting corrupted when editing the object vertices... I did clean up how mesh data is handled a little bit a while back, so maybe that's helping... Anyway it may still happen in some cases, but glad that its working better!

Closing this for now, Let me know if anything else comes up!

closed 2022-07-12 by kevancress