kevancress/MeasureIt_ARCH #225: More Area text placement oddness

baldyman01, 2022-07-13:
Blender 3.2.1 , MeasureIT_ARCH development 12/7/22

If I start a new file and add an area to the top face of the default cube the text comes out in the X-Y plane.
If I add a new cube and add an area to the top of that it comes out in the Y-Z plane and I can't find a way to change it. The view plane override was what I assumed should control that but it appears to change nothing.

[image]
[MeasureIT_ARCH Area Text Plane.blend.zip](https://github.com/kevancress/MeasureIt_ARCH/files/9103272/MeasureIT_ARCH.Area.Text.Plane.blend.zip)

kevancress, 2022-08-12:
Hey @baldyman01

Sorry for the slow reply. I started working on a fix for this in 1cf1c51abb03f1669fdee82f1b538f3493c0820a so if you grab the latest development version it should be slightly better.

Still some remaining weirdness with area dimension though... gonna take a better look at their draw code this weekend and try and get them tunned up

baldyman01, 2022-08-12:
Nice job - Areas are now centred and are aligned correctly even on complex objects like icospheres. Offset scales correctly with the object as well.

closed 2022-08-24 by kevancress