kevancress/MeasureIt_ARCH #227: Can't run image nor animation ?

CedricGhestem, 2022-07-26:
Hi 
Not that expert

Python: Traceback (most recent call last):
  File "C:\Users\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\MeasureIt_ARCH-development\measureit_arch_render.py", line 107, in execute
    outpath = render_main(self, context)
  File "C:\Users\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\MeasureIt_ARCH-development\measureit_arch_render.py", line 252, in render_main
    draw3d_loop(context, objlist)
  File "C:\Users\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\MeasureIt_ARCH-development\measureit_arch_geometry.py", line 3620, in draw3d_loop
    if not custom_call and not view.skip_instances:
AttributeError: 'NoneType' object has no attribute 'skip_instances'

kevancress, 2022-08-11:
Hey Cedric

This happens if you don't have a MeasureIt_ARCH view defined. Check the views settings in the Scene Properties to set one up.

I'll try and set a more informative error message for this

kevancress, 2022-08-11:
Added a catch that creates a default view if one is not set up, so that renders won't fail if no view is present.

980941bf806a5e63354f55e0c994c97325e77027

closed 2022-08-11 by kevancress