kevancress/MeasureIt_ARCH #229: File crashes while rendering

loonyproof, 2022-08-02:
crashes when pressed MeasureIt_ARCH image and MeasureIt_ARCH Animation

kevancress, 2022-08-11:
@loonyproof Please Indicate what version of MeasureIt_ARCH  and blender your using, any error messages that appear, and a set of steps to reproduce the crash... otherwise I can't do anything to solve this...

Re-open another issue with that information if your still having crashes. Thanks.

closed 2022-08-11 by kevancress

loonyproof, 2022-08-18:
sorry it took so long to reply(
Version MeasureIt_ARCH 0.4.7 and bledner 3.2.2.
Can you please tell me how to view the error report? Bledner crashes and there is no report in the console.
And for some reason, only some files crash, if you create a new file, then bledner does not crash.
Maybe the problem is that the files were created in an earlier version?

kevancress, 2022-08-18:
@loonyproof 

No Problem!

Version 0.4.7 had a lot of critical bugs that caused crashes, so not too surprised that its doing that.

First thing to try would be to download the latest development version of MeasureIt_ARCH from the main GitHub page and try that.

Error reports are visible by going to "window - > toggle system console" in Blender, it doesn't help too much on crashes, but it should list out which objects are being rendered, and if you watch closely sometimes you can see which object is causing the crash.

Let me know if updating to the lastet development version fixes things, if not I'll re-open this issue!

loonyproof, 2022-08-19:
Installed the addon from GitHub (MeasureIt_ARCH-development) and it worked!
Thank you!

loonyproof, 2022-08-19:
[image]
here the center moved out, can I fix it myself in the code (how?)?