kevancress/MeasureIt_ARCH #230: gizmo

rulon, 2022-08-09:
Is there any distance setting between "gizmo" arrow and according dimension?
I use arrow for drag drop
But at the moment, the margin from the corresponding measurement is too large.

[image]

kevancress, 2022-08-11:
@rulon Yeah this has been bugging me for a while too. It's just hardcoded right now, but I should make it proportional to the dimensions width so that it still works at small and large scales...

kevancress, 2022-08-11:
Alright so I added a basic scaling of the offset based on the dimensions length... e22fa1b5d3102cf39f966deb545f83bb4383606b

This should help, but isn't the best... Going to leave this issue open till I find a better solution

rulon, 2022-08-12:
Is it possible to make the dimension move when this text is dragged by mouse?

[image]

kevancress, 2022-08-12:
@rulon 

Yeah I think it should be possible, but would require a fairly significant overhaul of the Gizmo and dimension code...

I'll make a separate thread for it as a feature request so I can keep it in mind