kevancress/MeasureIt_ARCH #231: Line Group delete

ccalvoga, 2022-08-11:
Thanks for this increadible addon.
I realize that when you delete a Line Group the block of data is keeped. If you create a Line Group then you delete it, create a new one completely different to the previous one and change the name of this one to the deleted one, the new Line Group is the same as the deleted one.

kevancress, 2022-08-11:
Hey @ccalvoga

What version of the addon are you using. I think I fixed this already, so if you download and install the development version from the main github page that should fix this.

in either case, this is an issue with how line groups are cached for drawing on the gpu, tabbing into and out of edit mode on the object should fix it too

ccalvoga, 2022-08-19:
Hi @kevancress ,
I'm working with version 0.4.7. 
I have  tabbed into and out of edit mode on the object and still re-appear.
Here, there is video.

https://user-images.githubusercontent.com/40768363/185577240-553d42a3-76e5-4056-99cd-60ea1771776d.mp4

ccalvoga, 2022-08-19:
Hi Kevan,
I have dowloaded last release and it work perfect.
Thanks

kevancress, 2022-08-19:
Great thanks!

Let me know if  you find any other issues

closed 2022-08-19 by kevancress