kevancress/MeasureIt_ARCH #233: UPDATE DOCS!!!

kevancress, 2022-08-12:
Last time I did a documentation update was Feb 2021.... and I've added a ton of features since then. Need to do some proper documentation updates.

kevancress, 2022-08-12:
Making an outline here:

- [ ] Set Up GitHub pages
- [ ] Show Prev Commit Hash in Blender Ui (for Reporting)
- [ ] Add a copy system info button (for Reporting)
- [ ] Create Guidelines for reporting issues
- [ ] **Create Dedicated Docs Pages For:**
- [ ] Getting Started
- [ ] Dimensions
- [ ] Linegroups
- [ ] Annotations
- [ ] Styles
- [ ] Views
- [ ] Rendering
- [ ] Experimental Features (Schedules, Sheets, Etc)

kevancress, 2022-09-10:
Figured out how to host demo files on github pages, so I can add demo files to each tutorial as well !!!

ccalvoga, 2022-10-14:
Hi @kevancress,
I have try to use the Titleblock but I didn´t find the way. I have dowloaded the examplo listed in documentation but it seems to by empty. Could you please upload an example?
Thank a lot for you incredible work.
Carlos