kevancress/MeasureIt_ARCH #234: Hatching Pattern Size

baldyman01, 2022-08-12:


kevancress, 2022-08-24:
Hey @baldyman01 Think I've got the scale part of this sorted out in 9fe76a8a8d9eee729fa82a185689a62d3721a243. Give the latest development version a try and let me know how it works!

Still working on an offset option, and some way to make the svg pattern based on the hatch collections bounding box instead of the 1m x 1m positive X & Y direction. ( which should be a bit more intuitive when creating hatches)

baldyman01, 2022-08-25:
Hi Kev,

It took me longer than it should have to figure out that the Pattern size is in pixels.....

So to make sure I understand the logic before telling you it's wrong again -

An A4 view @ 150ppi is 1753x1240px - a Pattern Size of 1753 fits the view exactly width wise. 
[image]

The hatch does not currently scale with the drawing contents when View Settings -> Scale is changed. So 1:20 view produces -
[image]

Ideally you'd always want the pattern to exactly fill the width of the face it's applied to, but I can't see how you'd do that as it's completely arbitrary. In reality you'd normally be working with a few categories of wall each with a width and/or hatch.

So if you could take the pattern size from the hatch bounding box size it becomes related to the size of the mesh, not the size of the output/camera. All we'd have to do it set up one hatch per wall type, which is not exactly onerous.

I think I'm with you. Scaling does my head in.

kevancress, 2022-08-25:
Hey @baldyman01 

Yeah I think that's correct on the old behavior, but the change I pushed yesterday should take some of the calculation out of figuring that out!

The latest version should respect the true dimensions that the hatch was drawn at (accounting for resolution and scale automatically) So a grid hatch drawn with .25m spacing and a pattern size of 1 should render true-to-scale at .25m now.

having the pattern scale to match the face width / bounding box width would be super handy for insulation style hatches, but probably not something you'd want in all cases I'd imagine? Maybe an option in the hatch settings to "scale pattern size by face width" or something like that could work? If I can figure out how to do that haha.

Let me know if the new scaling works for you though. I think having it based on the scale it was drawn at, rather than in pixels, is probably easier to manage?

baldyman01, 2022-08-29:
Hi, Kev,
I'd managed to not update it properly. Yep the scaling looks right to me now.  It's much easier to manage now :)