kevancress/MeasureIt_ARCH #244: Duplicate elements

ccalvoga, 2022-09-19:
Hi Kevan,
It would be nice to add a bottom for copy parameter from a style to a new style. Same for Views. 
have a nice day,
Carlos

kevancress, 2022-09-23:
Hey @ccalvoga 

This actually already exists for views, its under the extras menu next to the list:

[image]

But your right I should definitely add this for styles as well!

kevancress, 2022-10-01:
Hey @ccalvoga

Just added a duplicate styles operator, its in the extras menu for styles

[image]

Closing this for now, let me know if there's any problems!

ca0d6a3fe4779b72c1ae585a7d62eec409fa07d3

closed 2022-10-01 by kevancress

ccalvoga, 2022-10-14:
Hi @kevancress ,
thanks a lot, 
It is really usfull when you have a lot of styles assinged to different views.

Carlos