kevancress/MeasureIt_ARCH #245: Is it possible to export the link map of vertices rendered in freetyle?

lisiyao21, 2022-09-28:
It is an amazing work!!!

I am a beginer user of blender and it is a black box to me. If I understand correctly, the freestyle is going to select suggestive edges from the whole view map (the mesh) and draw these edges to color. Is it possible to export/record those rendered edges/vertex info (including id, 2d position, 3d position...) for further processing?

Thank you!!

kevancress, 2022-09-29:
Hey @lisiyao21

This is probably a better question for the freestyle developers. MeasureIt_Arch can embed freestyle linework in its outputs but other than that it doesn't interact with freestyles view map data at all, so I'm not sure what data is available for export from freestyle.

closed 2022-10-03 by kevancress

lisiyao21, 2022-10-04:
Thank you for your kind reply!!!

lisiyao21, 2022-10-04:
So sorry for my late response due to hanging out...  Since I'm really new to blender, could you guide me which specific person should I consult? Thank you so much!!