kevancress/MeasureIt_ARCH #249: Font size for area does not work

giudansky, 2022-10-29:
Font size area style in scene properties, does not change fonts in the view.

kevancress, 2022-10-30:
Hey @giudansky 

This should be fixed now in that latest development version (commit 7642682f4745da8cbc2eb25de703ba912f70d072).

Give it a try and let me know if it works.

I'm going to leave this issue open, since right now styles for area dimensions are a bit of a hack, they're using the same settings as length dimensions, but should probably have their own style type.

giudansky, 2022-11-01:
Yes, it works now.
Not sure if it's still due to style type, but area overlays makes others lines/text negative, changing colors.
https://imgur.com/SGQ1EJa

Any way to make em black like all the others ?

kevancress, 2022-11-01:
Hey @giudansky 

hmmm that's odd. I can't seem to reproduce this on my end, could you post an example file that's having this issue?

giudansky, 2022-11-02:
Just found out it was due to the fill color in the "active dimension setting".

I love measureit arch. When will replace the old one inside blender ?

Thanks

kevancress, 2022-11-06:
Ahh okay, glad its resolved!

Thanks!!! No plans right now for it to replace MeasureIt. Actually, because MeasureIt_ARCH uses some external libraries for exporting to svg & dxf I'm not sure it would be accepted as a core addon that's included with Blender