kevancress/MeasureIt_ARCH #250: Title Block Demo file

baldyman01, 2022-11-09:
Hi Kev,

Hope you're doing well?

I've been fiddling with the Title Block option trying to see how it works and am coming up blank. It appears to do nothing at present so I'm probably missing something. I downloaded your Demo file from the documentation (great idea to have that BTW if documentation isn't available yet) but I appear to just have a blank Blender file.

MIA 0.5.0 + Blender 3.3.1 on Ubuntu 20.04

Cheers

Pete

kevancress, 2022-11-13:
Hey @baldyman01

Thanks for the reminder, I setup that demo file as a test to see if I could actually host files on GitHub Pages, but never actually filled it in.

Got a start writing the tutorial (not finished yet so its a bit sparse), but did actually put up a complete Titleblock demo file on that page.

https://kevancress.github.io/MeasureIt_ARCH/tutorials/titleblock_setup/

Take a look and let me know if you have any questions or if anything doesn't make sense!

Cheers!
Kev

baldyman01, 2022-11-16:
Hi Kev,
So I've got the title block working as I want it so thanks for that. The only oddity I've found is that if I move any part of the title block scene into +Z it no longer gets rendered in the main scene. Only things in -Z.  It renders normally in my title block scene camera. To double check it wasn't camera clipping distance causing the issue I moves the camera with the title objects and the +Z cuttoff still happens while maintaining camera Z distance.

I note that in your example blend everything in the title block is in -Z so you probably know about this little quirk already.
It's certainly not a show stopper but a note in the Docs might save some confusion for others.

kevancress, 2022-11-21:
Hey,

aahh yeah, the titleblock scene gets positioned with it's z axis aligned to the camera's local z axis, and the origin placed right at the camera's near clipping plane, so anything +Z in the titleblock scene gets cut off.

I'll add a note about that in the docs!