kevancress/MeasureIt_ARCH #252: Enabling fails with Blender 3.4.1

seichter, 2023-01-08:
- Arch Linux 
- Blender 3.4.1 from package manager

Latest head (today) does yield the following error when trying to enable the file:

```python
Traceback (most recent call last):
  File "/usr/share/blender/3.4/scripts/modules/addon_utils.py", line 333, in enable
    mod = __import__(module_name)
  File "/home/hartmut/.config/blender/3.4/scripts/addons/MeasureIt_ARCH-development/__init__.py", line 78, in <module>
    from . import measureit_arch_main
  File "/home/hartmut/.config/blender/3.4/scripts/addons/MeasureIt_ARCH-development/measureit_arch_main.py", line 33, in <module>
    from .measureit_arch_geometry import clear_batches, update_text, draw3d_loop, preview_dual
  File "/home/hartmut/.config/blender/3.4/scripts/addons/MeasureIt_ARCH-development/measureit_arch_geometry.py", line 75, in <module>
    aafrag = load_shader_str("aa_frag.glsl")
  File "/home/hartmut/.config/blender/3.4/scripts/addons/MeasureIt_ARCH-development/measureit_arch_utils.py", line 68, in load_shader_str
    shader_file = open(os.path.join(path, "glsl\{}".format(file)), "r")
FileNotFoundError: [Errno 2] No such file or directory: '/home/hartmut/.config/blender/3.4/scripts/addons/MeasureIt_ARCH-development/glsl\\aa_frag.glsl'
```

kevancress, 2023-01-09:
Hey @seichter

Thanks for reporting! I'd just been overhauling how the shaders get stored and loaded in the add-on, and had left some windows specific paths hardcoded in there, thanks for catching it!

Should be fixed in 0810e00f977dd36d3d8c3dd5ce1231dd18309d1a on the development branch, If you wouldn't mind testing it and seeing if enables on Linux now that would be a big help.

I actually do recommend people use the development branch, if they don't mind the occasional bug, It's been a while since I've done a proper update to the master branch so it's marginally more stable, but quite out of date.

kevancress, 2023-01-10:
Great! Thanks for confirming! Let me know if you have any other issues!

closed 2023-01-10 by kevancress