kevancress/MeasureIt_ARCH #253: RASTER RENDER IS BROKEN

kevancress, 2023-01-16:
Raster rendering is not working, for some reason it's just drawing the screen texture...

kevancress, 2023-01-16:
Looks like this has to do with changes to the gpu module in 3.4 + and not somthing I broke when messing with the draw pipeline... yay

vollstock, 2023-01-16:
On the mac, ist’s not only not rendering. The addon fails completely (won't even install).
Let me know, whether this is different on your machine and you want me to provide some Mac-specific info.

kevancress, 2023-01-16:
Hey @vollstock This is likely a separate problem from the GPU draw issues, so I've opened a separate thread for it: https://github.com/kevancress/MeasureIt_ARCH/issues/254 

It enables okay on windows so if you could let me know your Blender version, what version of the addon you downloaded, version of macOS, and any error messages that pop up when you try to enable the addon that would be great! Hopefully I can get that fixed today

kevancress, 2023-01-16:
Looks like the raster draw issue was just an issue reading frambuffers on the particular Blender 3.4 Alpha version I was using. 3.3, 3.41 stable, and 3.5 alpha don't have this issue. Closing for now

closed 2023-01-16 by kevancress