kevancress/MeasureIt_ARCH #254: MeasureIt_ARCH not enabling on macOS

kevancress, 2023-01-16:
On the mac, ist’s not only not rendering. The addon fails completely (won't even install).
Let me know, whether this is different on your machine and you want me to provide some Mac-specific info.

_Originally posted by @vollstock in https://github.com/kevancress/MeasureIt_ARCH/issues/253#issuecomment-1384194435_

vollstock, 2023-01-16:
hm, I'm not so sure anymore.
The latest git version doesn’t work in 3.3 either. An older Version does work in Blender 3.3 and 3.4.

The older version I have is not a git clone so I am having a hard time figuring out, what commit this is from.

vollstock, 2023-01-16:
just read you comment in the other issue.

Lates git doesn’t enable on Mac OS Blender 3.3.2, 3.4 and 3.4.1.
I can test a specific version if you had a suspicion what commit could have changed that?!

I will later check if I can get some info out of the log…

kevancress, 2023-01-16:
Ahh okay, Yeah I've been doing a lot of fiddling with the draw pipeline lately to get more dimensionally accurate lineweights, dashes etc.

Some changes to the way I was loading shaders caused some enable failures on Linux too, so I wouldn't be surprised if its a similar issue here. The log should point me right to the problem though!

kevancress, 2023-01-16:
going back to cc0ee72fe5c58360e3b5c4542e32e79c1d384227 should also resolve things, if I'm right about shader loading being the issue.

vollstock, 2023-01-16:
It seems, you are right:

```Python
M_ARCH import modules
ERROR (gpu.shader): pyGPUShader GeomShader: 
      | 
   65 | out flat int f_dashed;
      | 
      | Error: interpolation qualifier 'flat' must precede storage qualifiers
      | 
  225 |         f_dashed = verts[0].dashed;
      | 
      | Error: Use of undeclared identifier 'f_dashed'

Traceback (most recent call last):
  File "/Applications/Blender 3.4.1.app/Contents/Resources/3.4/scripts/modules/addon_utils.py", line 333, in enable
    mod = __import__(module_name)
  File "/Users/vollstock/Library/Application Support/Blender/3.4/scripts/addons/MeasureIt_ARCH/__init__.py", line 78, in <module>
    from . import measureit_arch_main
  File "/Users/vollstock/Library/Application Support/Blender/3.4/scripts/addons/MeasureIt_ARCH/measureit_arch_main.py", line 33, in <module>
    from .measureit_arch_geometry import clear_batches, update_text, draw3d_loop, preview_dual
  File "/Users/vollstock/Library/Application Support/Blender/3.4/scripts/addons/MeasureIt_ARCH/measureit_arch_geometry.py", line 117, in <module>
    allLinesShader = gpu.types.GPUShader(
Exception: Shader Compile Error, see console for more details
```

Will not check cc0ee72fe5c58360e3b5c4542e32e79c1d384227 …

vollstock, 2023-01-16:
yes, cc0ee72fe5c58360e3b5c4542e32e79c1d384227 works

kevancress, 2023-01-16:
Ahh okay, yeah macOS is generally stricter when compiling shaders so there's usually some bugs that don't show up on windows.

80498b5e5787b124f01e8d5aa8b0956beec9d057 should fix the compile error that you got above, give it a try and let me know if it works. There may still be other errors though.

Thanks!!

vollstock, 2023-01-16:
it unfortunatle doesn’t. Crashes immediately:

`/var/folders/pt/5v4qd7t97hq7v8h7kb7vs8fh0000gn/T/blender.crash.txt`

```md
# Blender 3.4.1, Commit date: 2022-12-20 08:50, Hash ef9ca44dee7f

# backtrace

# Python backtrace
  File "/Users/vollstock/Library/Application Support/Blender/3.4/scripts/addons/MeasureIt_ARCH/measureit_arch_geometry.py", line 114 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "/Users/vollstock/Library/Application Support/Blender/3.4/scripts/addons/MeasureIt_ARCH/measureit_arch_main.py", line 33 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1078 in _handle_fromlist
  File "/Users/vollstock/Library/Application Support/Blender/3.4/scripts/addons/MeasureIt_ARCH/__init__.py", line 78 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "/Applications/Blender 3.4.1.app/Contents/Resources/3.4/scripts/modules/addon_utils.py", line 333 in enable
  File "/Applications/Blender 3.4.1.app/Contents/Resources/3.4/scripts/startup/bl_operators/userpref.py", line 445 in execute
```

kevancress, 2023-01-17:
Hmm, well the good news is that "MeasureIt_ARCH/measureit_arch_geometry.py", line 114 is exactly where the shader that's causing problems gets loaded... so this is still likely a shader compilation issue. The bad news is we've got no specifics on what's causing the crash...

I'll do some digging today and see if I can track down anything that might be tripping up the shader compiler. If you get a chance, running blender from the command line with the "--debug-gpu" argument sometimes gives more detailed crash reports for this kind of issue, if you could try it out that would be a huge help!

Thanks!

vollstock, 2023-01-17:
yes, I figureds as much but wouldn’t know, what to test in the shader.

Sadly, when starting with debug-gpu flag Blender crashes right away, too.

```
~ % /Applications/Blender\ 3.4.1.app/Contents/MacOS/Blender --debug-gpu --factory-startup
WARN (gpu.debug): source/blender/gpu/opengl/gl_debug.cc:165 init_gl_callbacks: Failed to hook OpenGL debug callback. Use fallback debug layer.
zsh: segmentation fault  /Applications/Blender\ 3.4.1.app/Contents/MacOS/Blender --debug-gpu 
```

no idea why that happens or what to do about it :-/
(the --factory-startup flag doesn’t make a difference, I just thought, it might)

kevancress, 2023-01-17:
Strange. even with the add-on disabled it still crashes on startup with the "--debug-gpu"?

hmm well I'll still take a look through the shader today to see if I can catch what's causing the compile issue, hopefully I find something!

kevancress, 2023-01-17:
Alright, this may be a bit of a long shot, but I've cleaned up the shader code a bit in 3589f78b18404c641ebb68f01313b816307d02db and 10f99272e9c32879e4275fd45fb1bca82ff80a0f

Maybe that will fix the compile errors? if you could give the latest development version a test that would be great! Thanks!

vollstock, 2023-01-17:
Thanks but sorry, no good news. Same error, same line (although that’s 111 now).

I’ll file a bug report on developer.blender.org about the `--debug-gpu` flag.
I'm afraid, that however won’t produce a fast response. If you have any idea of what else I could do, I happily will.

vollstock, 2023-01-17:
ok, opened that [bug report](https://developer.blender.org/T103947).
While doing so, I noticed, that only Blender 3.4 won’t run with that flag, but 3.3 would.

It produced this error file:
[blender.crash.txt](https://github.com/kevancress/MeasureIt_ARCH/files/10439268/blender.crash.txt)

Don't know, if you can read anything from that. The only thing I noticed, is that the crash happend when compiling a shader. We knew that much already :-/

kevancress, 2023-01-23:
@vollstock

Sorry for the delay on this, been trying to track down / borrow a Mac so I can do some testing myself to try and solve this. 

Was looking back over the crash log today though and noticed that `glpLLVMGetFunctionGlobalVariableUse` shows up a fair bit right before the crash. I did have one spot in the geometry shader where I'm calling a function outside of the main shader function... I've removed that call in 9d34d0b5341c48b11b502fc381ab08b4f9042469 which might fix the problem...

Bit of a long shot still, but hopefully if that doesn't work I'll be able to get my hands on a Mac sometime this week to do some testing

vollstock, 2023-01-24:
Don't be sorry, you’re doing voluntary work. No expectations from my side.

Pulled latest commit and it now acivates without errors :-)
Texts and lines look different, but I guess that’s due to latest changes and has nothing to do with this issue.

Let me know, if you need me to check/test anything still.

kevancress, 2023-01-26:
Amazing! Glad that its working! I'll have to keep in mind that macOS doesn't support function calls outside of the main loop in glsl shaders.

Yeah I've been trying to make line weights and text more dimensionally accurate, actually defining lineweights and font sizes in pts (1/72nd of an in) to match other svg editors (inkscape, illustrator etc.) Its still not quite rendering properly for raster renders in perspective, but I've almost got that sorted out!

Sound good, I'll let you know if there's anything else to check! 

Gonna close this for now, but let me know if you find any other issues!