kevancress/MeasureIt_ARCH #258: Schedule export error

ChubbyQuark, 2023-01-24:
Getting the following error on Win 10 and OSX. I think it is the line of code in the screenshot as once this is removed the schedule exports as expected.

Support/Blender/3.4/scripts/addons/MeasureIt_ARCH-development/measureit_arch_schedules.py", line 235, in execute file_path = bpy.path.abspath(font_file) NameError: name 'font_file' is not defined

[image]

ChubbyQuark, 2023-01-26:
> Hey @ChubbyQuark <https://github.com/ChubbyQuark>
>
> Thanks for catching that! should be fixed in the latest development
> version e556399

Glad I could help.

The addon is amazing, just what Blender needs for designers and Architects.

As a side note, do you think it would be possible to view a tabulated data
in Blender (similar to how revit displays tabulated info.). Ive had a
couple of converstation with the guys at OSArch around the same thing, be
great to get your insight too.

All the best,
Chris.

kevancress, 2023-01-26:
Thanks! Glad people are finding it useful!

Yeah I've actually been working on a Table annotation object that reads data from a csv and creates a table in the 3D viewport. It's still a bit buggy, but the docs are here https://kevancress.github.io/MeasureIt_ARCH/prop_descriptions/tables/

And here's a quick demo too.

[image]

I wish I had more time to spend over in the OSArch forums! seems like there's always great things happening over there!

ChubbyQuark, 2023-01-27:
Wrong place for this but I don't have another contact for you, thought I'd
share the below:-

Spent a bit of time testing the new features on one of the larger models I
work with, the dynamic line group is great, really helps express details:

[image: image.jpeg]

kevancress, 2023-01-29:
Ooh interesting, I hadn't heard of Prop Chart, I'll have to check it out!

Glad that the dynamic line group is working well for you! if you ever need more control over it, you can filter what geometry it considers by creating a vertex group, and setting it in the lines 'Filter Group' property!

I'm gonna close this issue for now, but you can reach me by opening up new issues here, or by email at kevan.cress@gmail.com

Ps. the images on your last two comments didn't come through for some reason

closed 2023-01-29 by kevancress