kevancress/MeasureIt_ARCH #259: text is very low res

drewp, 2023-01-28:
[image]

My scene units are cm, and I don't think my scene is excessively large or small. 

blender 3.5 alpha from 2023-01-17

drewp, 2023-01-28:
Aha, I found the scene settings area, which is apparently like this by default: 
[image]

These let me dial to an acceptable res, though it would obviously be nice if the addon could come up with something more sensible in my case.

kevancress, 2023-01-29:
Hey @drewp

Yeah I've been trying to come up with a better balance for Text resolution! The main problem really is that the addons performs pretty slowly when drawing lots of medium-high resolution text, which isn't so bad at render time, but makes the viewport really sluggish (hence the lower default / preview resolution in the viewport). 

I might dial it up to 150 ppi by default, which looks a little cleaner, and doesn't hurt viewport performance too much... but ultimately I need to work on the text drawing performance to properly resolve this.