kevancress/MeasureIt_ARCH #260: Report or update your script to use 'gpu' module.

drewp, 2023-02-05:
When measurements are displayed, my .xsession-errors is flooded with these errors:
```

WARN (bgl): source/blender/python/generic/bgl.c:48 report_deprecated_call: 'bgl.glEnable' is deprecated and will be removed in Blender 3.7. Report or update your script to use 'gpu' module.
WARN (bgl): source/blender/python/generic/bgl.c:48 report_deprecated_call: 'bgl.glEnable' is deprecated and will be removed in Blender 3.7. Report or update your script to use 'gpu' module.
WARN (bgl): source/blender/python/generic/bgl.c:48 report_deprecated_call: 'bgl.glActiveTexture' is deprecated and will be removed in Blender 3.7. Report or update your script to use 'gpu' module.
WARN (bgl): source/blender/python/generic/bgl.c:48 report_deprecated_call: 'bgl.glBindTexture' is deprecated and will be removed in Blender 3.7. Report or update your script to use 'gpu' module.
```

This is in 3.5.0 alpha, so it won't affect most users for a while. It really affects me, though :)
```
✨ dash(pts/10):~% grep -a report_deprecated_call .xsession-errors | wc -c 
1856518226
✨ dash(pts/10):~% grep -av report_deprecated_call .xsession-errors | wc -c 
2327467
```

kevancress, 2023-02-05:
Hey @drewp

Yeah I've had this one on my radar for a bit. Blender is slowly phasing out the use of its openGL wrapper in favor of the new GPU module (I think this is all part of the migration to using Vulcan for rendering instead of openGL).

I've been tackling this in bits and pieces, but fixing it essentially requires a complete re-write of all the addon's drawing code, so it may take a while.

Not sure if there's a way to suppress/quiet these errors until then.

drewp, 2023-02-05:
SGTM. IMHO blender should use a warn_once type of function for these. It's not useful to get a billion of them.

kevancress, 2023-02-06:
Hey @drewp 

I've started to do some of the migration to the new gpu api in this commit, 0061327e7fb8b9c9cc7e41b84c236c11e36642ef

I'm sure it won't remove all the warnings your seeing but should start to reduce some of them

kevancress, 2023-02-07:
Took on removing the bgl calls from the text drawing system in aff07ffe609829eac5c41caf8e1392b44f000a32

As a side effect this has actually improved the performance of drawing text, so I've rasised the default preview resolution, which should resolve https://github.com/kevancress/MeasureIt_ARCH/issues/259

drewp, 2023-02-15:
Installed a github zip from just now. (and a fresh 3.5.0 alpha too) Still some warnings, but nothing that repeats per-frame :)

reloading addon: MeasureIt_ARCH-development 1674881006.6449885 1676433332.1527555 '/home/drewp/.config/blender/3.5/scripts/addons/MeasureIt_ARCH-development                                 /__init__.py'
Modules Installed () from '/home/drewp/Downloads/MeasureIt_ARCH-development (1).zip' into '/home/drewp/.config/blender/3.5/scripts/addons'
WARN (bgl): source/blender/python/generic/bgl.c:54 report_deprecated_call: 'bgl.glEnable' is deprecated and will be removed in Blender 3.7. Report or update your script to use 'gpu' module.
WARN (bgl): source/blender/python/generic/bgl.c:54 report_deprecated_call: 'bgl.glEnable' is deprecated and will be removed in Blender 3.7. Report or update your script to use 'gpu' module.
WARN (bgl): source/blender/python/generic/bgl.c:54 report_deprecated_call: 'bgl.glBlendFunc' is deprecated and will be removed in Blender 3.7. Report or update your script to use 'gpu' module.
WARN (bgl): source/blender/python/generic/bgl.c:54 report_deprecated_call: 'bgl.glDepthFunc' is deprecated and will be removed in Blender 3.7. Report or update your script to use 'gpu' module.
WARN (bgl): source/blender/python/generic/bgl.c:54 report_deprecated_call: 'bgl.glEnable' is deprecated and will be removed in Blender 3.7. Report or update your script to use 'gpu' module.
WARN (bgl): source/blender/python/generic/bgl.c:54 report_deprecated_call: 'bgl.glDepthFunc' is deprecated and will be removed in Blender 3.7. Report or update your script to use 'gpu' module.
WARN (bgl): source/blender/python/generic/bgl.c:54 report_deprecated_call: 'bgl.glEnable' is deprecated and will be removed in Blender 3.7. Report or update your script to use 'gpu' module.
WARN (bgl): source/blender/python/generic/bgl.c:54 report_deprecated_call: 'bgl.glEnable' is deprecated and will be removed in Blender 3.7. Report or update your script to use 'gpu' module.
WARN (bg
etc

kevancress, 2023-12-17:
Just finished moving all the drawing code over to the gpu API now, so these warnings should all be gone now and things should be working on Blender 4.0

closed 2023-12-17 by kevancress